LaserWeb/LaserWeb4 #93: UI Enhancement: Prompt for name and path when saving G-Code

gndprx, 2016-12-19:
It seems to be the default to save any files as gcode.gcode in the system configured download folder. 

Could we please prompt for the path and name of the file after clicking the Save G-Code button to cut some time out of the process?

tbfleming, 2016-12-19:
* chrome://settings/
* Show advanced settings…
* Turn on: Ask where to save each file before downloading

funinthefalls, 2016-12-19:
Simple, thank you. LaserWeb is getting so well designed that it causes a lot of us to forget that it is running in a standard browser.

openhardwarecoza, 2016-12-19:
Filesaver.js allows one to specify the filename
https://github.com/eligrey/FileSaver.js in LW1 we saved the output with the
same name as the input file (although in lw2/3/4 which loads multiple files
at a time it isn't elegant anymore) -  what about a quick modal to ask for
file name?

openhardwarecoza, 2016-12-19:
> eligrey/FileSaver.js in LW1 we saved the output with the same name as the

In LW3 we had an invokeSaveAsDialog()  function
https://github.com/LaserWeb/LaserWeb3/blob/master/public/js/main.js#L469
you can reuse?

openhardwarecoza, 2016-12-19:
> In LW3 we had an invokeSaveAsDialog()  function https://github.com/
> LaserWeb/LaserWeb3/blob/master/public/js/main.js#L469 you can reuse?
>
> On Dec 19, 2016 8:33 PM, "Peter van der Walt (Gmail)" <

Nb: invokeSaveAsDialog used native os dialog to let you enter filename

tbfleming, 2016-12-19:
> what about a quick modal to ask for file name?

Then we'll get two modals: one from us and one from the browser.

openhardwarecoza, 2016-12-19:
Agreed,  modal isnt cool.  That's why i went to check what we do in lw3:
invokeSaveAsDialog()  doesn't use a modal

gndprx, 2016-12-19:
The first option that Todd posted works for me. I honestly forgot you could override that in the browser settings :(

I'll go ahead and close this one unless anyone has something more to add and keep this in mind when we get around to adding documentation to the new site that Peter is working on.

gndprx, 2016-12-19:
Closing as this can be configured via the browser and does not require a software change.

closed 2016-12-19 by gndprx