LaserWeb/LaserWeb4 #94: Make the sizing / coordinate bubble vertical

cojarbi, 2016-12-19:
I use tab to navigate ( dont know if many out there do ) so if i want to change lets say the coordinates i need to click X change the value then tab 5 times to get to Y.

I could just double click the field to highlight it, then delete or type the new value but Tab makes more sense , at least to me. 

Anyone else thinks the same?

Or maybe re - arrange the way the tabs works?

iceblu3710, 2016-12-27:
Indeed, Min(x,y) Center(x,y), I also enter coordinates in X,Y pairs instead of modifing all X's then the Y's.

Should just be a matter of changing the tab order if this was HTML/CSS... I see `class FloatingControls extends React.Component` and that the DOM is rendered in `render()` but I have no idea how to change the tab order in there.

tbfleming, 2016-12-27:
@iceblu3710 try adding ```tabIndex``` (note the case) attributes to the inputs in the following code:
```
<tr>
    <td><span className="label label-danger">X</span></td>
    <td><Input value={round(bounds.x1)} onChangeValue={this.setMinX} type="number" step="any" /></td>
    <td><Input value={round((bounds.x1 + bounds.x2) / 2)} onChangeValue={this.setCenterX} type="number" step="any" /></td>
    <td><Input value={round(bounds.x2)} type="number" onChangeValue={this.setMaxX} step="any" /></td>
    <td><Input value={round(bounds.x2 - bounds.x1)} type="number" onChangeValue={this.setSizeX} step="any" /></td>
    <td rowSpan={2}>
        &#x2511;<br /><input type="checkbox" checked={this.state.linkScale} onChange={this.linkScaleChanged} /><br />&#x2519;
</td>
</tr>
<tr>
    <td><span className="label label-success">Y</span></td>
    <td><Input value={round(bounds.y1)} onChangeValue={this.setMinY} type="number" step="any" /></td>
    <td><Input value={round((bounds.y1 + bounds.y2) / 2)} onChangeValue={this.setCenterY} type="number" step="any" /></td>
    <td><Input value={round(bounds.y2)} type="number" onChangeValue={this.setMaxY} step="any" /></td>
    <td><Input value={round(bounds.y2 - bounds.y1)} type="number" onChangeValue={this.setSizeY} step="any" /></td>
</tr>
```

iceblu3710, 2016-12-27:
WOW, I glazed right over the bloody obvious table render in `class FloatingControls `, I'm really out of practice.

Is there a xml, yaml, or json settings file to store changes for mundane things like this? A client_settings  file I can put FloatingControls.alt_tabIndex = true into?

If not I recommended this format `feature name, description, options, current setting` That way you can easily add a user config feature without hard coding it into the settings panel, just parse and display the client_settings.json.

tbfleming, 2016-12-27:
Don't add a setting for this; too many settings annoys users. e.g. users flocked from Mozilla to Firefox partly because Firefox stripped out most of Mozilla's settings.

tbfleming, 2016-12-27:
I suspect most users will prefer the new (X Min, Y Min, ...) tab order.

iceblu3710, 2016-12-27:
Hmm, good point. 

I should branch this project now that I am getting into the guts. Took a day to figure out the organization but now I think I see how it's fitting together. The live dev environment is new to me but pretty darn handy.

iceblu3710, 2016-12-27:
#106

iceblu3710, 2016-12-27:
Pull Request #106  to add the functionality that @cojarbi requested.

tbfleming, 2016-12-27:
It's now on the live site.

cojarbi, 2016-12-27:
Thanks for this @iceblu3710

cojarbi, 2016-12-27:
It works as required just a question regarding the navigation. Should it keep the cursor trapped and the user clicks to get out of it or like now that one's it reaches the last field it goes to the next field on the screen/navigation. Reason is maybe the users needs to do some other changes that are affected by a change on the first field. Example I locate the path on the origin and then change the size.

iceblu3710, 2016-12-27:
I thought of it looping in the popup, but seeing how the popup currently shows up whenever selecting objects that would mean hijacking the tabbing functions across the whole application.

The easiest way is to use the standard tab & shift-tab to navigate the popup.

cojarbi, 2016-12-27:
Understood. Hijacking the navigation  is not what we want so unless someone has another suggestion I think we can close this one

closed 2016-12-28 by cojarbi