LaserWeb/LaserWeb4 #95: Default common settings for operations by machine

gndprx, 2016-12-20:
Many of the CAM operations require entering the same details over and over in the various settings. As part of the machine setup, can we provide a list of operations to allow you to set the default values and then apply these when you select the operation in the CAM panel?

I know it will vary from material to material, but if a user uses 3mm ply 51% of the time, it would benefit them to setup the defaults for that and during the times I use other materials I can simply change the settings as needed at CAM time.

This would be hugely beneficial to both the laser and cnc work streams and is common in some form or another in many CAM tools.

tbfleming, 2016-12-20:
See #39

gndprx, 2016-12-20:
Yup, you're absolutely right. 

Closing as a duplicate of #39

closed 2016-12-20 by gndprx