LaserWeb/LaserWeb4 #96: 3D Viewer: Support Opacity as S value

openhardwarecoza, 2016-12-20:
[image]

Works in LW3, needed in LW4:

Note: Opacity scals between 0-100% opaque, based on the S value but S differs between various firmwares so kep in mind to read scale from settings

openhardwarecoza, 2016-12-20:
LW3: https://github.com/LaserWeb/LaserWeb3/blob/master/public/lib/gcode-viewer/gcode-parser.js#L299-L306

tbfleming, 2016-12-20:
Might need a different approach that doesn't mess up the laser preview. Turn off gcode and documents but leave laser on for that example. Zoom in and out.

openhardwarecoza, 2016-12-20:
[image]

Aha! Thats what I was looking for! thick lines (0.1mm bam, 0.5mm stepover to show effect above)  makes zoomed in view look better than lw3! Just need to document hiding gcode...

tbfleming, 2016-12-20:
Also try with diagonal and the Sim slider :)

openhardwarecoza, 2016-12-20:
blown away! See thats why you need to post screen2gifs in development news again!

closed 2016-12-20 by openhardwarecoza

tbfleming, 2016-12-20:
Been too busy coding. I think I did screen2gifs of laser preview a while back.

openhardwarecoza, 2016-12-20:
you did yes... will post some today (; keep on coding!