LaserWeb/LaserWeb4 #101: Inconsistencies on Gcode generation (Open Paths on closed shapes)

jorgerobles, 2016-12-22:
This is a big Gif, so I put in GDrive https://drive.google.com/drive/folders/0B9GOB0erdDVbTTIyRkEyYUlSNW8?usp=sharing

The Gif shows how shapes (the same) are generated some times closed, some times not.
Also, closed paths gcode as open paths.

I've also noticed Hi CPU consumption, using 100% CPU (OSX 10.9, GTX 760 (CUDA enabled), Chrome 55.x)

tbfleming, 2016-12-23:
The closed paths issue is fixed. I'm looking into the GPU issue.

tbfleming, 2016-12-23:
GPU per-frame stats:
* 9178 triangles to fill polygons
  * Normally ok; GPUs can go beyond this.
  * But, there's a lot of overlap because of the overlapping polys; this can tax the GPU pipeline at the fragment level.
* 11174 line segments for polygons: ok; GPUs can go beyond this.
* 21036 triangles for thick selection lines. Might be an issue.

Top 6 profiler results (self time):
* 11.24%: uniformMatrix4fv. Very odd; this WebGL function shouldn't rank high in the profile.
* 10.16%: draw (in regl). This doesn't include time in WebGL functions. Also very odd. Ideally this function would just speed time in WebGL.
* 9.51%: bufferData. This WebGL function handles data transfers; it normally ranks high.
* 9.25%: draw (another function with the same name, in regl).
* 8.71%: "(program)". Don't know what this is.
* 6.04%: toString. Called by regl. Why does regl need this?

There's also a lot of time spent collecting garbage. I'm thinking of dumping regl and going directly to WebGL; the time in uniformMatrix4fv and the two draw functions isn't good. It would also help me keep garbage under control.

tbfleming, 2016-12-27:
I dropped regl for native WebGL and reoptimized rendering. @jorgerobles please retest it to if it helps with this file on your machine.

jorgerobles, 2016-12-27:
Done. It's working beautifully. Superb job!

jorgerobles, 2016-12-28:
Perfomance issues arise again. How do I to generate the stats you logged?

tbfleming, 2016-12-28:
I wrote temp code (no longer exists) to report the per frame stats. The profiler generated the rest: https://developers.google.com/web/tools/chrome-devtools/rendering-tools/js-execution

tbfleming, 2016-12-28:
It looks like you changed the file. Does the performance differ between the two file versions?

jorgerobles, 2016-12-28:
Yes, the file was changed to improve visualization (I wrote a wiki page with tips to people that use Adobe AI). I'm trying to determine what makes it slow down, if its related with file structure or something else: grouping, fills or simply paths amount. I think the file I'm using is complex enough to being a good test bench.

The performance was horrible before your patch and improved after ditching regl, when the appearance of the paths was not parsed (using css styles).
Now the file has appearance correctly represented (using attributes) and the performance have drop (quite but not as bad as the first time)

If we can keep the code you wrote to test several files maybe we can get the weak point. I could be enabled/disabled under settings/tools, or so.

tbfleming, 2016-12-28:
Please attach the file that uses attributes here.

tbfleming, 2016-12-28:
Here's how to profile. The percentages under Self Time are what matter most. The time in ms depends on how long you take the same.

[image]

jorgerobles, 2016-12-28:
[escaleras_a4_300.svg.zip](https://github.com/LaserWeb/LaserWeb4/files/676185/escaleras_a4_300.svg.zip)

Thanks!

tbfleming, 2016-12-28:
That file works ok on my machine. See what the profiler shows on yours.

[image]

jorgerobles, 2016-12-29:
[CPU-20161229T100550.cpuprofile.zip](https://github.com/LaserWeb/LaserWeb4/files/676842/CPU-20161229T100550.cpuprofile.zip)

[image]

[image]

As far as I see, choking occurs upon mouse events (on timeline).
But general interaction is slowdown when "documents" are rendered in canvas (toggling visibility). Any other combination (Gcode+Laser) does not incurr in that. I suppose is normal, though. 

I'm gonna try with different OS (mavericks is a bit old)

jorgerobles, 2016-12-29:
A notable optimization. The file is much more manageable right now 👍 👍

closed 2016-12-31 by tbfleming