LaserWeb/LaserWeb4 #103: CAM UI: Using buttons (alternatively) instead of dragging; naming operations

jorgerobles, 2016-12-24:
Hi, working with a lot of paths (like the one posted on #101 ) I've encountered drag operations is somewhat a difficult task. So I've added (not comitted yet) two buttons in the drop area.

Add Single. Adds all selected documents as a sigle operation (just like now)
Add Each. Adds each selected documents as an operation (so there will be an operation for each selected object)

Maybe objections about space could arise, as all of this cramp ever more the cam pane, but I assure you this is a very productive way ('cause a side issue is that a canvas selected shape is not highlighted on side pane if it's inside a group)

Naming operations should be good too. I ended Up with a dozen of Laser Cut operations in the pane, and have to iterate each of them to search what shape is in (for discarding a shape inside)

[image]

tbfleming, 2016-12-24:
Go ahead and push this. Maybe call them "Create Single" and "Create Multiple"?

tbfleming, 2016-12-24:
@openhardwarecoza I now agree after seeing several people struggle with D&D.

jorgerobles, 2016-12-24:
I will do ;)

openhardwarecoza, 2016-12-24:
(: no problem @tbfleming UIs evolve from user complaints (;

tbfleming, 2016-12-31:
Operations now have names.

closed 2017-01-06 by jorgerobles