LaserWeb/LaserWeb4 #105: Drag & Drop On Linux

iceblu3710, 2016-12-27:
Trying out the latest build and none of the drag and drop features work. 

Eg, I load a file in the cam tab and I can see it appear on the grid and its associated features are listed. I can't however, drag it to the gcode area. When I try and drag the file it just highlights the text.

Tried Chrome, Firefox and different Node versions.

System:
```
Chrome: Version 54.0.2840.100 (64-bit)
Firefox: 50.1.0
OS: Ubuntu 16.04.1 LTS
Kernel: 4.4.0-57-generic
npm: 3.5.2 & 3.10.9
node: v4.2.6 & v6.9.2
```

tbfleming, 2016-12-27:
LW4 only supports Chrome 55. Chrome 55 added pointer events, which unifies mouse, touch, and pen handling. We have no plans to support Firefox.

iceblu3710, 2016-12-27:
I will upgrade my Chrome and make sure that's the problem.

I only use firefox to make sure page rendering errors are not a coding error.

iceblu3710, 2016-12-27:
Chrome 55 works as expected. Some SVG errors but that's for a different issue.

closed 2016-12-29 by tbfleming