LaserWeb/LaserWeb4 #108: Cannot select inner shape.

jorgerobles, 2016-12-28:
Cannot pick inner shape. I suppose collision detection is taking hollow shapes as filled ones. A workaround could be working with filled shapes to reorder shape front-back, but will be very annoying working with (visualization
[image]
)

[escaleras_a4_300.svg.zip](https://github.com/LaserWeb/LaserWeb4/files/676185/escaleras_a4_300.svg.zip)

@tbfleming I've been experiencing performance issues again (reopened #101)

tbfleming, 2016-12-28:
I switched it to only hit test on path outlines when paths aren't filled.

jorgerobles, 2016-12-28:
Tested And Perfect, thank you!

closed 2016-12-28 by jorgerobles