LaserWeb/LaserWeb4 #109: Save the current state of file and operations

cojarbi, 2016-12-28:
When generating operations for complex files with multiple path there is a point where it could be beneficial to be able to save the current state of the file itself and the operation applied to that file / paths, for earthier further review or to change any other setting without having to redo everything.

I think its a very attractive feature to have

tbfleming, 2016-12-29:
Right now this is buried under Settings -> Tools -> Download Snapshot. Check the Documents + Settings + Operations checkboxes first.

@jorgerobles could you add "Save" and "Load" buttons to the CAM pane? "Download" "Upload" "Store" and "Recover" are confusing names.

cojarbi, 2016-12-29:
Whats the difference between Download and Store?

I downloaded a snapshot ( documents+settings+operations) but upload does not do/show anything.

jorgerobles, 2016-12-29:
Duplicate from #33 (for reference)

@cojarbi, @tbfleming  I've done a complete revision on Tools panel, better visual cues and legibility.
I've added a catch for LS errors due overquota, too.

About Load/Save buttons on CAM, what would it be, to localstorage, to file, or both?
Where would you like to place them, top or bottom?

@tbfleming do you know any reliable way to gzip this data? A complete Json dump (like the "escaleras" file) goes up to 15mb.

tbfleming, 2016-12-29:
Just file. I prefer plain json to aid debugging.

cojarbi, 2016-12-29:
Sorry about the duplicate. Maybe we can close this one and continue on 33

jorgerobles, 2016-12-29:
No problem. Was just a note.

cojarbi, 2016-12-29:
@jorgerobles no worried just saw it was closed. I'm unable to use this feature on Windows or Mac. 
This is what im doing: 
Load my file and operations
Go to settings select what I want in my snapshot. Settings, file, 
Clicked download

Then I refresh the browser , go to settings and click load snapshot. Nothing happens ( also did select  items to load with no effect )

Am I missing something or what?

jorgerobles, 2016-12-29:
@cojarbi Please take a look https://drive.google.com/open?id=0B9GOB0erdDVbT1NoSVlfRXB0U28
Gif is a bit shaky, sorry.

Basically, the workspace is composed of the state Operation, CurrentOperation, and Documents
Settings store the machine settings.

I'm going to finish the cam load/save toolbar to the workflow will be better.

cojarbi, 2016-12-29:
OK i see. The difference is I did not take current operations just operations. Is the settings option the same as the above alone?

⁣Ariel Yahni​

jorgerobles, 2016-12-29:
Yes, it is.

jorgerobles, 2016-12-29:
Done, Load/Save now appear on the top of CAM pane

cojarbi, 2016-12-29:
It that live?

tbfleming, 2016-12-29:
It is now.

cojarbi, 2016-12-29:
Working beautiful , thanks so much. so this will save file and operations only correct?

openhardwarecoza, 2016-12-29:
> @jorgerobles <https://github.com/jorgerobles> here's the command line I
> use to make changes go live:
>
> git checkout gh-pages && git pull && git merge dev-es6 && npm run bundle-dev && git add dist && git commit -m regen && git push && git checkout dev-es6

(: add that to package.json
Could be 'npm run golive' (;

jorgerobles, 2016-12-29:
@tbfleming, sweet!

jorgerobles, 2016-12-29:
Hmm not for me, I have not set browser config option to "save as". I will investigate if that config is accessible to avoid double modal.

jorgerobles, 2016-12-30:
Cannot access to chrome://settings from javascript (to detect saveAs mode), so cannot avoid duplicated save file modals :frowning_face:

tbfleming, 2016-12-30:
The browser's modal is much better (includes folder navigation). We could remove our modal and suggest people change the option. We can probably force the option on Electron.

funinthefalls, 2016-12-31:
I LOVE the save operations feature! I can be working on a file., saver it, log out, then log in later and continue working on it later right from where I left off. I believe lot of people will enjoy this feature as well, so please continue to develop it!

openhardwarecoza, 2016-12-31:
Might even be an idea to offer to save on exit (; if you close the browser,
 save state to local storage,  on startup, load last state?

funinthefalls, 2016-12-31:
Option to auto save would be a fantastic feature, add it to the settings to enable or disable it. As well, can we grab the original file name when bringing docs into CAM so vectors etc would be named with the filename/element id sort of like MyPlaqueVector3 or MyPlaqueImage5 when working with a file named MyPlaque. Or should that be brought up as a feature request in a new issue?

jorgerobles, 2016-12-31:
About AutoSave, there's a problem with localstorage, the ~5mb limit could
be reached easily, and not a single browser has the same space assigned to
it.
As far as I know, there's no way to get the space available nor remaining
on browser only an exception if limit has reached.
Said this, the feature could be done with some limitations: Actually,
without backend (node server, electron) could try to save upon exit, and if
an exception arise, prompt user to exit without saving current workspace.
Upon backend development, the state could be sent to save a
"last_session_snapshot.json"

openhardwarecoza, 2017-01-13:
Working well.  Closing for now.  May open new issue later when node server is ready to autostore to backend

closed 2017-01-13 by openhardwarecoza