LaserWeb/LaserWeb4 #114: Touch screen optimisation

FabCreator, 2017-01-03:
Hi all, 

We use a Pipo X9s to run the interface for the FabKit, however i think there are many users looking to run tablet devices.

Looking at laserweb 4 a few of the features i can see being really hard to use on a touch screen device. 
However Laserweb 4 is more complex in terms of features and operations so running the whole program on a touchscreen will not really the best solution.

Perhaps it would be possible to create a "mach3/pronterface" version of laserweb just to load gcode and send to the machine have options to adjust speed and power meanwhile after your laptop or desktop has sent the gcode to the "host" is free to disconnect or do somthing else and is not tethered to the laser. etc 

From a development point of view it may help too as the separation of G code generation and coms means less potential conflict. (i would guess)

looking at it currently many users are already using lw3 to run the gcode made in lw4 so perhaps this is the way to go?

cprezzi, 2017-01-03:
LaserWeb (3 & 4) uses a node server for the communication with the machine. The only thing that needs to stay connected to the machine (after a job is started) is this node server.

If you install the node server on a Raspberry Pi, you can disconnect you client (tablet) after starting the job, and the node server will finish it. You can also reconnect the client to pause or stop the running job. However, I don't know if a Raspi is fast enough.

tbfleming, 2017-01-03:
Maybe the node server could run on the X9 and the PC could connect to it

FabCreator, 2017-01-03:
@cprezzi thats cool, we have the X9 bascily because the pi is a little slow at times. we also think its important to have an interface on the machine, and from a saftey point of view you should not be able to operate your machine remotely/without someone there.

@tbfleming thats a good idea, Would it be possible to have the jogging and general control buttons and maybe a local safe and load for past gcodes so you dont have to always set it up agian if its a frequent job.

On the computer side of things there is a simple push to nodesever/X9 on the computer running laserweb 3/4

openhardwarecoza, 2017-02-28:
I'll drop my +1 in this issue - fingering the small buttons are quite hard (:

FabCreator, 2017-03-07:
@openhardwarecoza would it be possible to just make a Laserweb driver. Optimised for touch screens just simple jogging command line and load g code. remove the preview aswell. make it super light?

emteeoh, 2017-03-07:
Two comments:

1) maybe we should make a "mobile" interface for phones and tablets?
2) I'm running lw3 on a pi3 and I have yet to notice how it's slower than running from my laptop. Although, I don't use rasters much. 

Sent from a something something...

jorgerobles, 2017-03-07:
Hmm I think is a bit overkill to do some computations on tablets. At the
end, its running JavaScript, not an optimized c piece of software.

 Anyways light ui will be a whole subproject. The responsive css is not a
real mobile UI

funinthefalls, 2017-03-07:
For a light interface, why not use LW on a PC to generate the gcode and send it to the Laser using pronterface

openhardwarecoza, 2017-03-07:
See the lw.comm-server project. Already pipelines and communicated

jorgerobles, 2017-03-07:
@funinthefalls then we need to move computations to lwserver

funinthefalls, 2017-03-07:
OK, gotcha ya.

openhardwarecoza, 2017-03-07:
Wtf @funintefalls - Claudio worked way to hard. Why would you recommend pronterface. Disgraceful

funinthefalls, 2017-03-07:
Was not meant to be disrespectful. My apologies if taken that way.

jorgerobles, 2017-03-07:
Was a joke ;)

openhardwarecoza, 2017-03-07:
@jorgerobles probably wasnt. Anthony is away a lot and often passes out ill informed advice like that.

jorgerobles, 2017-03-07:
Ouch. Zipmouth myself.

FabCreator, 2017-03-07:
@openhardwarecoza so a separate sender (laserweb pronterface for example) just needs a UI and could tie into the coms from @cprezzi ?

openhardwarecoza, 2017-03-07:
No, lw.comm-server will have a minimal interface built in. Its in the works

openhardwarecoza, 2017-03-07:
@FabCreator 

See https://github.com/LaserWeb/lw.comm-server/issues/6#issuecomment-282105618 - it states: 
"I also planned to convert the lw3 testpage later to a responsive mini control panel, so users can additionally connect with a mobile or tablet to see how far the job is and pause or stop it." - from Claudio

And aa couple other places where I dont have the links right away

cprezzi, 2017-03-07:
@FabCreator Yes, lw.comm-server can be used from other frontends too. They just need to implement our API. I tried to hide all interface or firmware specific stuff in the server and created a universal API.

Additionally, I have planned to integrate a slim web frontend for remote control from a mobile or tablet, as Peter told.