LaserWeb/LaserWeb4 #115: Pocket operation not working

cojarbi, 2017-01-04:
Attached gcode for a pocket operation; it goes into initial position , plunges but then it seems not to move. I can hear some noise coming from the steppers as its trying to move but feels more like a shake
[gcode (37) copy.gcode.txt](https://github.com/LaserWeb/LaserWeb4/files/683706/gcode.37.copy.gcode.txt)

tbfleming, 2017-01-04:
Can you attach the workspace file?

cojarbi, 2017-01-04:
Here you go. Mind that the above gcode is only the mill pocket operation on the attached workspace
[iTopie Final (4) copy.json.txt](https://github.com/LaserWeb/LaserWeb4/files/683801/iTopie.Final.4.copy.json.txt)

tbfleming, 2017-01-04:
The firmware was probably choking on the tiny moves. I reduced the resolution for milling to .001 mm.

See what happens now. What firmware are you using?

cojarbi, 2017-01-04:
This is with grbl 1.1e

cojarbi, 2017-01-04:
Did a clear cache, reloaded the workspace, generated code only for the pocket and the result is the same. 
It moves on the decimals of only one coordinate. I can see the gcode has bein reduced to 3 decimal places
[gcode (23) copy.gcode.txt](https://github.com/LaserWeb/LaserWeb4/files/684630/gcode.23.copy.gcode.txt)

tbfleming, 2017-01-04:
Mill operations now have a ```segment``` option (mm). The CAM trims moves less than this length. You'll have to play with this value to get a good tradeoff between accuracy and speed.

cojarbi, 2017-01-04:
Ok thanks  will test later as I don't need super accuracy here at the moment

cojarbi, 2017-01-05:
Used a segment of 0.5 and its seems to be working fine for this project.

cojarbi, 2017-01-05:
I can say this is solved, thanks

closed 2017-01-05 by cojarbi