LaserWeb/LaserWeb4 #116: Curve within curve exclusion

DarklyLabs, 2017-01-05:
Is there a current or planned method to exclude the region of a curve within another curve in fill operations?

For instance, the inside two curves in the letter 'B'.

DarklyLabs, 2017-01-05:
I cannot see this option in the 'Laser Fill Path' settings.

tbfleming, 2017-01-05:
Most operations need a clear definition of what's inside and what's outside; combining paths gives them that definition. "Laser Cut" doesn't need a clear definition, so it's optional.

DarklyLabs, 2017-01-05:
@tbfleming How would we define the inner and outer curves in the letter 'B' for instance?

Other software we have worked with made an assumption that inner curves would be subtracted from the fill operation. This assumption is generally what most users would want to do and makes it very straight forward to use. If they didn't want this, then they would not include inner curves into their fill operation.

tbfleming, 2017-01-05:
A single path object (e.g. a B) may have holes and disconnected regions. 'combine paths' does a union operation between multiple path objects.

jscut has a dropdown to select between union, difference, intersection, and xor. I didn't include it because it seemed to cause confusion. Instead I encourage people to create proper path objects using other tools.

tbfleming, 2017-01-05:
I could switch it from union to xor. That will cover both the disconnected and the hole cases well. It may cause confusion where there's partial overlap, which is why I went with union.

DarklyLabs, 2017-01-05:
Based on what most of our users would be doing would be handled by xor (if that means excluding the inner curve region).
If there is overlap between curves then that usually means users have to fix this problem in their designs as a 'fill' operation would not be able to be defined correctly.

tbfleming, 2017-01-14:
This change is now live.

DarklyLabs, 2017-01-14:
@tbfleming Just checked it. Works beautifully. Thank you.

closed 2017-01-14 by openhardwarecoza