LaserWeb/LaserWeb4 #117: Remote control

kabcpc, 2017-01-10:
Greetings!
I want to make a remote control interface for LaserWeb. I'm going to use webSocket (esp8266 serer).

I want to get a recommendation from the developer. How best to organize the management remote control interface?

cprezzi, 2017-01-10:
The lw.comm-server (not finished yet) will be able to have more than one client connected. This way you can run your PC with the normal web frontend for prepairing and starting the job, but also can controll the state of the running job (pause/resume/stop) from a second client like your esp8266. You just need to connect the same server and use the corresponding WebSocket commands.
At the moment, there is no documentation of the WebSocket interface, but I will publish that when the server is ready.

openhardwarecoza, 2017-01-10:
To elaborate: https://github.com/LaserWeb/lw.comm-server will be the
serial/server backend for LaserWeb4
(unlike LaserWeb3 where everything was in one repository)

closed 2017-01-13 by openhardwarecoza