LaserWeb/LaserWeb4 #121: DXF not loading: filetype issue

openhardwarecoza, 2017-01-12:
I heavily depend on the guitarlist sketchup dxf export in my workflow https://github.com/LaserWeb/LaserWeb3/wiki/Workflow:-Sketchup-2D

In LW3 we just look at file extension, there these work. 
In LW4 this doesnt: (note the console.log should print file.type, and its absense makes me think we arent identifying it as dxf?)
[image]

Test file:

[end-caps.dxf.txt](https://github.com/LaserWeb/LaserWeb4/files/702422/end-caps.dxf.txt)

openhardwarecoza, 2017-01-12:
Logged the action.payload

[image]

iceblu3710, 2017-01-13:
dxf-parser outputs the dxfTree with 10 entities. They are `color: 16777215` which is white.

[image]

[image]

iceblu3710, 2017-01-13:
Oh, might be a problem I had after installing DraftSight. If you check the mime type of the dxf it should be `image/vnd.dxf` as the document load process checks the file extension in cam.js to choose a parser then off to the document reducer which reads the mime-type.

After install DraftSight my dxf's where `application/vnd.VAR_VENDOR_NAME.VAR_PRODUCT_NAME-dxf type (application/vnd.VAR_VENDOR_NAME.VAR_PRODUCT_NAME-dxf)` and I had to rebuild my mime types (I'm on Ubuntu)

openhardwarecoza, 2017-01-13:
Yes,  but we cant force users to fix files via Draftsight first.  Please
install LaserWeb3,  and play with it a little (: -  dxfs of all kinds
worked really well there.  All that didn't work was splines.

iceblu3710, 2017-01-13:
@openhardwarecoza My suggestions were to figure out if the dxf was not loading for you because of a mime type mismatch on your system. The file loads fine for me with a new git checkout so I can not reproduce the error.

tbfleming, 2017-01-13:
The mime type comes up empty on my machine also.

iceblu3710, 2017-01-13:
@tbfleming My screen cap is right from the `document/cam.js` at the dispatch call `dispatch(loadDocument(file, dxfTree));`

As a whole, Why do we check the extension in the documents and then the type in the reducer? It would be a lot easier to just use extensions. Mime types are pretty unreliable, DraftSight messed mine up but its not like the file actually changed or anything.

openhardwarecoza, 2017-01-13:
"Often, the MIME type sent by the browser will be the one reported by the
OS. And this is exactly why, as stated in the question, the MIME type
reported by the browser is unreliable."
So because you have Draftsight installed,  your OS reports it.  I have QCad
and sketchup (all i will ever use) which clearly then doesnt.  When it
comes to users...  We have to handle that unpredictability.

My suggestion:

if payload.file.type == dxf || file.name == *.dxf {} (simplified of course)

openhardwarecoza, 2017-01-13:
`It would be a lot easier to just use extensions` -  amen (:

tbfleming, 2017-01-13:
The mime types for images and svg files are reliable. Only switch DXF loading to check for extensions.

closed 2017-01-13 by tbfleming

openhardwarecoza, 2017-01-13:
> Closed #121 <https://github.com/LaserWeb/LaserWeb4/issues/121> via #123

Thanks!

PR #123 (2017-01-13): dxf's load by extension only
Fixes #121
commit 80585e0a (2017-01-13): dxf's load by extension only

Fixes #121