LaserWeb/LaserWeb4 #127: Need bullseye feature

cprezzi, 2017-01-15:
I am integrating comms module into LW4 and need the bullseye feature we had in LW3 to show the actual position. Is this feature allready in or can somebody make it available please?

jorgerobles, 2017-01-15:
Bullseye? do you mean the Jogs G92 button?

openhardwarecoza, 2017-01-15:
@jorgerobles: its a object in the 3D viewer that shows on the grid,  where
the DRO pos is:

Updated on every DRO change:
https://github.com/LaserWeb/LaserWeb3/blob/master/public/js/viewer.js#L27-L38

And initially drawn here
https://github.com/LaserWeb/LaserWeb3/blob/master/public/js/viewer.js#L251-L281

Used to be a 'bullseye'  in later versions we made it a cone (but the name
stuck)

jorgerobles, 2017-01-15:
> js/viewer.js#L27-L38
> js/viewer.js#L251-L281

Oook, sorry, name was a bit confusing.

openhardwarecoza, 2017-01-15:
> issuecomment-272704102

(: yip sorry.  Thats why i explained the history too. Hehe.  Feel free to
rename

tbfleming, 2017-01-15:
Added. I put an example in com.js showing how to set it.

cprezzi, 2017-01-16:
Perfect thanks!

closed 2017-01-17 by openhardwarecoza