LaserWeb/LaserWeb4 #128: Gcode Preview

iceblu3710, 2017-01-15:
I think G0 and G1 should be previewed on screen in different colors, say red for G1 and green for G0?

tbfleming, 2017-01-15:
They are.

iceblu3710, 2017-01-15:
Not for me. just did a pull and both are just green.

[image]

tbfleming, 2017-01-15:
Can you attach your gcode file?

[image]

iceblu3710, 2017-01-15:
[gcode.gcode.txt](https://github.com/LaserWeb/LaserWeb4/files/706815/gcode.gcode.txt)

tbfleming, 2017-01-15:
Very odd. This is with your settings file:

[image]

iceblu3710, 2017-01-15:
What's the last commit on your working version? 
`048bd1f 2017-01-15 | #127: add work pos marker (HEAD -> dev-es6, origin/dev-es6, origin/HEAD) [Todd Fleming]`

I don't get different colors with any file type or importing settings.

iceblu3710, 2017-01-15:
Just tried to reset my settings to the GRBL default and then the initial G0 move did not show up at all and everything was still red.

[gcode (1).gcode.txt](https://github.com/LaserWeb/LaserWeb4/files/706846/gcode.1.gcode.txt)
[laserweb-snapshot (1).json.txt](https://github.com/LaserWeb/LaserWeb4/files/706847/laserweb-snapshot.1.json.txt)
[laserweb-settings (1).json.txt](https://github.com/LaserWeb/LaserWeb4/files/706848/laserweb-settings.1.json.txt)

tbfleming, 2017-01-15:
I'm also on 048bd1f.

The (1) files also work on my machine. Which OS and Chrome version are you on?

I'm on Windows 10 with Chrome 55.0.2883.87 m (64-bit).

iceblu3710, 2017-01-15:
Not working on:
Ubuntu Linux 4.4.0-57-generic Chrome Version 55.0.2883.87 (64-bit)

Just tested on my HTPC and its working:
Windows 10 with Chrome Version 55.0.2883.87 m (64-bit)

iceblu3710, 2017-01-15:
There are reports of the `m` sub version causing things to act differently. Searching github issues for the `m` version shows 661 open issues.

Maybe it's in the browsers canvas rending or something?

iceblu3710, 2017-01-15:
Will wait for a Linux Chrome update. Closed for now.

closed 2017-01-15 by iceblu3710