LaserWeb/LaserWeb4 #129: Laser preview not rendering

iceblu3710, 2017-01-15:
Laser preview does not render on any cam functions.

tbfleming, 2017-01-15:
It should. Please save your workspace and your settings (2 different files) and attach them.

iceblu3710, 2017-01-15:
[laserweb-settings.json.txt](https://github.com/LaserWeb/LaserWeb4/files/706809/laserweb-settings.json.txt)
[laserweb-snapshot.json.txt](https://github.com/LaserWeb/LaserWeb4/files/706810/laserweb-snapshot.json.txt)

tbfleming, 2017-01-15:
This is with your settings and snapshot:

[image]

It draws the laser at the width in the settings, which is thinner than the gcode preview. For it to show up you need to turn off both Show Gcode and Show Documents.

iceblu3710, 2017-01-15:
I don't get any laser preview.

In the workspace json `laserDiameter": 0,` I changed to `laserDiameter": 1.0,` and increased the `"machineBeamDiameter": 0.2,` in the settings json.

Once imported I still do not get any laser preview.

openhardwarecoza, 2017-01-15:
Beam Dia in Settings tab?

iceblu3710, 2017-01-15:
Increased to 1.0 as well and no changes

iceblu3710, 2017-01-15:
Problem traced to a minor version difference in Chrome between Linux and Windows.

Not working on:
Ubuntu Linux 4.4.0-57-generic Chrome Version 55.0.2883.87 (64-bit)

Working on:
Windows 10 with Chrome Version 55.0.2883.87 m (64-bit)

Will wait for a Linux Chrome update. Closed for now.

closed 2017-01-15 by iceblu3710