LaserWeb/LaserWeb4 #131: Add a Drill operation

cojarbi, 2017-01-17:
I'm not sure what would be needed here besides final depth, plunge. I think bit size is/should be assumed?

Searched all open issue not finding a duplicate

tbfleming, 2017-01-17:
Found it. It was #22. I think @openhardwarecoza closed it because it also requested unrelated ops and wanted canned cycles, something firmware doesn't support.

To not ruin drill bits it would need various parameters defining the peck drill cycle. It would have to emulate the canned cycle using a series of G0, G1 vertical moves.

openhardwarecoza, 2017-01-27:
Yip, indeed why it was closed (:
Not really supported by any of the firmwares yet

closed 2017-01-27 by openhardwarecoza

arthurwolf, 2017-01-27:
http://smoothieware.org/drillingcycles