LaserWeb/LaserWeb4 #132: Set initial size of raster images by "pixel size" = "laser diameter" and allow changes by %

cprezzi, 2017-01-26:
Instead of definig a default dpi setting for raster images, it would be easyer to just use the "laser diameter" setting for the initial pixel size. The user could then change the picture size by a % value (insted of a non transparent dpi value).

What do you think @openhardwarecoza, @tbfleming, @lautr3k...

jorgerobles, 2017-01-26:
The only issue I could foresee is that a large image could go out of canvas

tbfleming, 2017-01-26:
After the initial load, the user sets the desired size in mm.

cprezzi, 2017-01-26:
@tbfleming : Even better.

jorgerobles, 2017-01-30:
Is this definitive? Should we ditch raster dpi setting and use px/beamDiameter instead, or should keep raster DPI and enable a "use beam size relation" toggle?

DarklyLabs, 2017-01-30:
My vote would be to define the the final image size required and the laser step over. This seems the most intuitive and generally follows traditional cnc terminology.

Linking engraving to beam diameter has its difficulties (especially with diodes) as the resultant engraved diameter for a pixel could change with power.

Being able to define an engraving 300x200mm in size and a step over of 0.15mm seems easy for everyone to understand. The laser moves horizontally along and then steps down 0.15mm for the next line.

tbfleming, 2017-01-30:
@DarklyLabs that's what we currently do. The desired size in in the floating controls. The not-quite step over is named "Laser Diameter" in the Operation config. "Not quite" because lw.rasterizer doesn't have a constant step over; it varies but stays close to the Laser Diameter value. I suspect it's rounding to nearest pixel center point.

cprezzi, 2017-01-30:
I think what @DarklyLabs means is that the term "Laser Diameter" is not automatically understud as "pixel size". Probably we should call it "Raster Pixel distance" or "Raster spacing"?

What I was meanig is to use that "Raster spacing" only for the initial load of a raster image (instead of some dpi setting). After that, the user can set the wished size with the existing floating controls.

tbfleming, 2017-01-30:
Since the rasterizer resamples the image, maybe we should avoid terms like dots, pixels, DPI, resolution, etc. All those could refer to before or after the resampling, causing confusion. Maybe replace the initial DPI setting with initial width.

jorgerobles, 2017-01-31:
@cprezzi sorry but then I misunderstood the meaning of the issue. 
Here you are talking about lw-rasterizer operation parameters, and not the DPI settings in settings pane. Isn't it?

I think, the DPI on pane settings is fine (graphic arts standard). I find rasterizer parameters naming confusing.

cprezzi, 2017-01-31:
@tbfleming : By scaling a raster image to a fixed size or dpi on load we potentially loose resolution and sharpness. That's why I'd like to scale it by default to the resolution of the rasterizer output, which is laser diameter. 

@jorgerobles : Yes, I am talking about raster images only, not vector. But the DPI setting for raster images is not optimal, as raster images (fotos) don't realy have a dpi, they have a certain pixel resolution and no size. By initially scaling the image width to pixel with * laser diameter we have a 1:1 pixel to rasterizer output scaling, which delivers the best resolution possible. This way, we could just drop the raster dpi setting.

cprezzi, 2017-01-31:
Additionally I would replace the term "Laser Diameter" by something clearer like "Raster Spacing" or "Raster Pixel Size". That would also make it clearer for operations like CNC-engraving of infills.

cprezzi, 2017-01-31:
@DarklyLabs The term "step over distance" would be correct, but only people with milling background understand that term (yet). We anyways have to give an explanation on mouse-over.

jorgerobles, 2017-01-31:
Ok, then I figured out well first time :) 
Scenarios:
a) I've got a 500x500px wide photo, and 0.2 beam diameter on machineSettings. Then, you propose the at the initial load, size should be 100mm on canvas, as is the optimal size to raster, isn't it? 

b) Then comes a photoshop guy, with a 150x100 mm photo he scanned at 300dpi. When he loads on laserweb, the image will be (1334x1181px) => 354mm wide.
(The photoshop guy will be annoyed, nothing special, all photoshop guys are picky by nature.)

Maybe a "use optimal size" button in raster operation could solve this...

tbfleming, 2017-01-31:
I suspect most people want to make the raster fit their stock, not the other way around.

cprezzi, 2017-01-31:
@jorgerobles Yes, a) is exactly what I wanted.
b) The photoshop guy should be familar with scaling caused by different machine resolutions (dpi's). Basically, by setting the Laser Diameter we set the machine dpi ;)

@tbfleming Setting an initial load size would probably cause more problems. 
For example: a user sets 300mm with as the default size, because he uses 300x200mm playwood. Then he loads a logo with 400x300 pixel. He will get a completely blurred engraving.

I think it's better to autoscale it for an optimal result (good for our reputation). If he scales the picture up, it's his decision (and his own fault).

cprezzi, 2017-01-31:
@jorgerobles We could probably use a "raster scaling" dropdown: 
* scale to optimal size
* scale to fixed width

jorgerobles, 2017-01-31:
I find "scale to optimal size" button enough inside the "raster operation", so we have the case b by default, as now is, and give a easy chance to change to the optimal size.
or maybe we could add it to the object size floating panel?
Doing a) is trivial (I did a one line test) anyways.

DarklyLabs, 2017-01-31:
I agree that keeping the aspect ratio of the image is important.

If the user imports an image that is 400x300pixels and they want to engrave it onto material that is 200mm wide, then the resultant engraving should be 200x150mm.

If their photo resolution is too low and they want to make a large engraving, then their engraving will be blurry. Users will understand this.

The important parameter is the user being able to choose the engraving size based on their material size, as tbfleming mentioned.

cprezzi, 2017-01-31:
@DarklyLabs Automatically changing the aspect ratio was not the discussion. Just scaling.

DarklyLabs, 2017-01-31:
@cprezzi Sorry, I misunderstood.

jorgerobles, 2017-02-23:
@DarklyLabs, @cprezzi  There's a new set of buttons on FloatingControls. Allow to scale to optimized for raster and revert to original

cprezzi, 2017-02-23:
@jorgerobles Perfect.
It would also be nice, if the anker point of the picture (on scaling) would follow the setting of auto positioning (like bottom left). 
If I want to place my pictures on bottom left, I want them to stay there even when I'm scaling them.

cprezzi, 2017-02-23:
At the moment, the anker point is allways the center of the picture.

jorgerobles, 2017-02-23:
Ok, I will figure out :)

jorgerobles, 2017-02-23:
@cprezzi, done!

cprezzi, 2017-02-23:
@jorgerobles You are the king!

DarklyLabs, 2017-02-23:
@jorgerobles Excellent! This feature makes it so much easier than having to work all that out.
(Now all we need is the ability to rotate!)

closed 2017-02-28 by openhardwarecoza