LaserWeb/LaserWeb4 #133: Load initialState defaults (operation reducer, among others)

jorgerobles, 2017-02-01:
The initialState of the "operation" reducer  does not hold a way to setting defaults from the settings store. Example:

All the Laser operations should load some defaults (laserDiameter) in state.
There's no way to set default values since initialState is a plain dictionary upon state creation thus the store is not created.

jorgerobles, 2017-02-01:
@tbfleming please check this, thanks!

tbfleming, 2017-02-01:
It will work, but I'm uncomfortable with reducers accessing GlobalStore. An alternative: add ```laserDiameter``` to ```dispatch(addOperation({...```.

jorgerobles, 2017-02-01:
It's not very DRY for me (Yes is an escape hatch, dug up redux docs and found nothing more academic). As far as GlobalStore is just for reading, no race conditions could arise.

Indeed is not a reducer accessing the store, is an action dispatching the attributes (stored on a reducer file yes :smile:, could move it to the actions/operation, but found reducer file closer to centralize the "model" state). 

Anyway If I find some better I will take use, for sure.

closed 2017-02-02 by jorgerobles