LaserWeb/LaserWeb4 #134: Air Assist

DarklyLabs, 2017-02-02:
Feature required to indicate if air-assist is installed and whether it needs to be activated during an operation.

Machine Settings -> 
-Air-Assist Installed [Checkbox]
-- Air-Assist ON command [M0X)
-- Air-Assist OFF command [M0X]

During operations, if air-assist checkbox is selected then an option to use air-assist during that operation should be made available.

If control over air-assist on a per-operation basis is too tricky, then a global control for enabling air assist when gcode is generated will suffice.

openhardwarecoza, 2017-02-02:
It might not be elegant, but how about utilising the Machine Profiles:  Create two profiles: One for "with air assist" and one for "without air assist" - and then have different gcode in the Start/End GCODE fields to enable air at start of job, and disable at end of job?

jorgerobles, 2017-02-02:
What would be an standard gcode for air assist ( M7, M8, M9)? Just to know.
Also, could you bake a simple gcode file to reflect the desired behavior?

Pooky1306, 2017-02-02:
Should it be more generic for "devices-that-need-to-run-during-job" including chiller, spindle cooling system, exhaust fan, whatever, etc ?

@jorgerobles smoothie use Mx commands assigned to switch pinX. You define the pin number and the on and off commands  (like M41>on, M42>off).
I don't know the GRBL side of the force ;)

tbfleming, 2017-02-02:
If you have air assist, don't you always want to use it? In that case, start and end gcode handle it.

jorgerobles, 2017-02-02:
@Pooky1306  Yes of course, indeed is planed like "whatever ON" and "whatever OFF".
But want to know if should be done at "start job/end job" level, or "every layer/path" or "this operation only" :smile:

@tbfleming, that was my question :)

openhardwarecoza, 2017-02-02:
Grbl is M08/M09

openhardwarecoza, 2017-02-02:
@tbfleming that was my suggestion too, have seperate machine profiles for
with / without - each profile has different start/end gcode

jorgerobles, 2017-02-02:
Right now, I've got it almost done (UI, options etc).
I think they need at operation level, so each laser operation have a "use air assist", being enabled by default on all operations if is enabled on settings, and you could disable at operation level if needed.
 Just need to figure where to put the ON and OFF on gcode generation :)

FabCreator, 2017-02-02:
On the FabKit M106 and M107 turn on and off the compressor, it is also possible to adjust the power of the compressor. 

Chiller i think is best left out of it as you should run it constantly not only when cutting, allowing your tube to warm up and then rapidly cooling is bad for its life.

A way to set the speed/power and on/off compressor and exhaust fan might be a nice feature to have in the UI and when setting up the job. no need to have the fan and compressor running at full when engraving but when it gets to the cutting you want it to increase automatically, could be an additional feature when setting up the operations.

Why would you not have a air compressor installed? everything is better with a compressor. (Well maybe not the tranquillity of the workshop XD)

cprezzi, 2017-02-02:
@jorgerobles I don't see when not to use air-assist, if I have it.  

A Lasersaur for esample activates air-assist with job start and deactivates it a few seconds after the job is finished. 

Something like `M8\nG4P2` for start and `G4P2\nM9` at end should be enought.

jorgerobles, 2017-02-02:
Thanks, I suggest wait for @DarklyLabs if they need this behavior or could be simplified (as they opened the issue)

> During operations, if air-assist checkbox is selected then **an option to use air-assist during that operation** should be made available.

DarklyLabs, 2017-02-02:
@jorgerobles We use M08 and M09 to start and stop the air-assist.
The desired behaviour would be:

> ; Pass Z Height 10mm (Offset: 0mm)
> 
> **M08       ; Air Assist ON**
> G1 Z10.00
> 
> ; Pass 0 Path 0
> G0 X8.89 Y0.00
> G1 X9.42 Y0.04 S1.00 F1000
> ....
> ....
> **M09       ; Air assist OFF**
> M2          ; End

DarklyLabs, 2017-02-02:
There are a number of situations where air assist is not desirable to be used.
The obvious one is when cutting very thin and delicate materials. Paper, fabric, foils etc will neither benefit or remain in place or un-deformed with air assist on.
The noise factor is the other consideration.

DarklyLabs, 2017-02-02:
As I mentioned originally, if control over air-assist on a per-operation basis is too tricky, then a global control for enabling air assist when gcode is generated will suffice.
Switching between profiles seems a little clumsy, especially since air-assist is a common accessory for laser cutters, possibly more so than a rotary axis, which has an activation feature in LW4 already.

jorgerobles, 2017-02-02:
Done :)  Please take a look to ensure is behaving as required.

DarklyLabs, 2017-02-02:
@jorgerobles Settings tab does not open up for us and no air assist options appear in the operations.
We updated to latest LW4 and cleared cache.

DarklyLabs, 2017-02-03:
@jorgerobles Ok, just ran some more testing. Looks like the 'settings' tab appears on our windows machine but not on any of our Mac machines (very strange).

On the windows machine, the air assist is working exactly as expected. GCode is correctly generated.

Pooky1306, 2017-02-03:
Settings tab doesn't open for me too (windows7)

jorgerobles, 2017-02-03:
@DarklyLabs , @Pooky1306 Fixed! was a bug with webcam, from other commit.(https://github.com/LaserWeb/LaserWeb4/commit/611c7cca83cb66feb101352b14842f94ecb850d4)

closed 2017-02-04 by jorgerobles