LaserWeb/LaserWeb4 #136: If external Gcode contains G20 then Scale by x25.4 onwards

openhardwarecoza, 2017-02-02:
So we have a couple of whiney Imperialists...

When they try and load external gcode made by other CAMs (nevermind that LWCAM is BEST CAM) their G20 (Inches) gcode renders 25.4x too small 

Can we please extend the gcode parser / preview to accomodate G20 gcode? (Ie preview inches) - as it the machine controller will execute it correctly, and the preview is just cosmetic that it appears too small, but oh well, lets humour them, we'll convince the next generation that Metric is Great

tbfleming, 2017-02-02:
The temp parser can't handle most gcode. G20 is just the tip of the iceberg.

openhardwarecoza, 2017-02-02:
@kaosat-dev (;

kaosat-dev, 2017-02-02:
I thought my work was redundant and that the temp parser was not temp anymore, but ok I'll gladly jump back in

openhardwarecoza, 2017-02-28:
Closing this for now (:  @kaosat-dev 's life is crazy at the moment - will revisit this someday when he is sorted first! (;   
Good luck sir!

closed 2017-02-28 by openhardwarecoza