LaserWeb/LaserWeb4 #139: Using DrawCommands

tbfleming, 2017-02-03:


tbfleming, 2017-02-03:
Quick overview; assumes you're familiar with ReGL.

For the basic template, look at ```basic.js```. It shows how to handle uniforms and a single attribute (positions). When you have multiple attributes, they have to be stuffed in a single array buffer; see ```thick.lines.js```. ```image.js``` shows how to handle image samplers; its ```texture``` argument comes from ```DrawCommands.createTexture```.

Frame buffers allow you to draw to textures; ```DrawCommands.createFrameBuffer``` creates a frame buffer and a texture to go with it. ```DrawCommands.useFrameBuffer``` redirects all drawing to the texture that belongs to the frame buffer object.

If you create any objects, then you need to ```.destroy()``` those objects when you no longer need them. It's usually better to create an object and reuse it than to frequently create and destroy objects.

jorgerobles, 2017-02-04:
Ok, I will take a look with clean mind :)

jorgerobles, 2017-02-14:
Hi, well, wonder to make some tests to get camera working as an image document in the workspace (I will figure out later if this is desirable), I found the place to start @ workspace.js:236

Questions:
- REGL automatically takes DOM video element and constructs as texture. What will do the same in LWGL?
- What is the purpose of cachedDocument?
- Should the be the video image cached? the video is streaming images so the Texture changes every frame.

tbfleming, 2017-02-14:
Don't use a document. Instead add the video to the draw() lambda in setCanvas() in workspace.js.

* Try ```texture.set({image: videoElement})```. This should grab the current frame from videoElement. Untested.
* Don't use cachedDocument. For its purpose, see the comment above DocumentCacheHolder.
* No

jorgerobles, 2017-02-14:
> Try texture.set({image: videoElement}). This should grab the current frame from videoElement. Untested.

Sorry, but where? Should I create the texture before isn't it?
Where is the place to create the texture?

tbfleming, 2017-02-14:
A good spot to create the texture is in setCanvas right after it creates hitTestFrameBuffer. The draw callback is called once per frame. Before the call to this.grid.draw() is a good spot to fill the texture, process it, and draw the result.

jorgerobles, 2017-02-14:
I've just figured out :)

In REGL, I've used several framebuffers to dump the results, as I have no idea how to do on a single framebuffer.

How can I achieve this on LWGL?

tbfleming, 2017-02-14:
Multiple framebuffers is probably easiest. Create your framebuffers where it creates hitTestFrameBuffer.

jorgerobles, 2017-02-14:
how do I dump the result framebuffer to the canvas? using image texture = framebuffer texture?

tbfleming, 2017-02-14:
The easiest method is to use drawCommands.image() and pass it the framebuffer's texture member.

jorgerobles, 2017-02-20:
@tbfleming what's the way to translate this line https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/draw-commands/webcam.js#L118 from REGL to LWGL?

now I have 
<pre>
let data = drawCommands.createBuffer(new Float32Array([0, 1, 2],[0, 2, 3],[2, 1, 0],[3, 2, 0]));
    return ({lens, fov, texture }) => {
        drawCommands.execute({
            program,
            primitive: drawCommands.gl.TRIANGLES,
            uniforms: { uLens:lens, uFov: fov, uSampler: texture },
            buffer: {
                data,
                stride: 0,
                offset: 0,
                count: 4,
            },
        });
    };
</pre>

tbfleming, 2017-02-20:
LWGL doesn't support elements, so you'll have to expand the data array by hand. 0-3 in the original are indexes into https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/draw-commands/webcam.js#L100 and https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/draw-commands/webcam.js#L106

After expansion, you should end up with 12 vertexes. Each vertex will contain aVertexPosition.x, aVertexPosition.y, aVertexPosition.z, aTextureCoord.x, aTextureCoord.x.

jorgerobles, 2017-02-20:
Sorry.. I did not understand. :slightly_frowning_face: 
<pre>
elements:
[0, 1, 2],[0, 2, 3],[2, 1, 0],[3, 2, 0]

vertexCoords:
-1.0, -1.0, 0.0,
1.0, -1.0, 0.0,
 1.0, 1.0, 0.0,
 -1.0, 1.0, 0.0

textureCoords:
 0.0, 0.0, 1.0, 0.0, 1.0, 1.0,  0.0, 1.0
</pre>

So it will be ...
<pre>
let data = drawCommands.createBuffer(new Float32Array(
[-1.0, -1.0, 0.0, 0.0, 0.0],//index 0
[1.0, -1.0, 0.0, 1.0, 0.0],//index 1
...
));

</pre>

jorgerobles, 2017-02-20:
I've end up with 

<pre>
let data = drawCommands.createBuffer(new Float32Array(
        [-1.0, -1.0, 0.0, 0.0, 0.0 ], [1.0, -1.0, 0.0, 1.0, 0.0], [1.0, 1.0, 0.0,  1.0, 1.0],
        [-1.0, -1.0, 0.0, 0.0, 0.0 ], [1.0, 1.0, 0.0,  1.0, 1.0], [-1.0, 1.0, 0.0, 0.0, 1.0],
        [1.0, 1.0, 0.0,  1.0, 1.0], [1.0, -1.0, 0.0, 1.0, 0.0], [-1.0, -1.0, 0.0, 0.0, 0.0 ], 
        [-1.0, 1.0, 0.0, 0.0, 1.0], [1.0, 1.0, 0.0,  1.0, 1.0], [-1.0, -1.0, 0.0, 0.0, 0.0 ]
    ));
    return ({lens, fov, texture }) => {
        drawCommands.execute({
            program,
            primitive: drawCommands.gl.TRIANGLES,
            uniforms: { uLens: lens.map(parseFloat), uFov: fov.map(parseFloat), uSampler: texture },
            buffer: {
                data,
                stride: 0,
                offset: 0,
                count: 4,
            },
        });
    };

</pre>

and a bunch of 
`.Offscreen-For-WebGL-0x7fef9595da00]GL ERROR :GL_INVALID_OPERATION : glDrawArrays: attempt to access out of range vertices in attribute 0`

tbfleming, 2017-02-20:
data: looks right, except it has to be a 1-d array.
stride: # of bytes in each vertex: 20
offset: 0. This field lets you skip data at the beginning of data.
count: # of vertexes: 12

jorgerobles, 2017-02-20:
It works!!!!
why stride=20,  4 bytes for 32bit x 5 positions?

One more thing. the image appears flipped in vertical. I will prefer if every "filter" flips the output.
How can I achieve that?

And thanks!

tbfleming, 2017-02-20:
>stride=20, 4 bytes for 32bit x 5 positions?

yes

flipped vertical: change:
```
return glCoord  * vec2(1.0, -1.0)/ 2.0 + vec2(0.5, 0.5);
```
to
```
return glCoord  * vec2(1.0, 1.0)/ 2.0 + vec2(0.5, 0.5);
```

jorgerobles, 2017-02-20:
Well we have some progress, but I cannot get a full image of the video past barrelDistort processing. It's overscaled and changing the size values did not work.

[image]

I tried to mimic the way I have in regl piping thru https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/draw-commands/webcam.js#L2 but I don't get any workable result.

jorgerobles, 2017-02-20:
Seems to be related with the workspace size. If panes are moved, the image change offsets  O_o

openhardwarecoza, 2017-03-05:
If I understand correctly we can close this issue (we can draw, its the effect stack that is still todo?) (aka continued from last update in #30?

jorgerobles, 2017-03-05:
Yes, this issue is related to documentation of use of lwgl. The main
stopper is that is not only lwgl (as a rewrite, abstraction inspired on
regl) but the use of webgl.

Thus the work of Todd is awesome in this area I miss the use of a down to
earth library as threejs, just to less gifted devs like me.

So close it.

<https://github.com/LaserWeb/LaserWeb4/issues/30>?

closed 2017-03-05 by openhardwarecoza