LaserWeb/LaserWeb4 #140: Raster beam power range

DarklyLabs, 2017-02-04:
lw.raster-to-gcode.js includes the ability to set the beam power range which is not included in LW4's rastering operation.

[image]

This is very important for achieving the correct intensity and desired contrast for different materials.

Can this be exposed in the raster operation options?

jorgerobles, 2017-02-04:
@DarklyLabs It's exposed but with split in settings and operations.

Beam Range the S range (on settings/gcode) 0-255 (0-65500)
The beam power already exposed in operation.(https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/lib/cam-gcode.js#L114)

:)

DarklyLabs, 2017-02-04:
@jorgerobles How can we set a beam power of 25-65%?
I don't see that ability in the operation.

DarklyLabs, 2017-02-04:
@jorgerobles Ok, I understand how its setup now.
The most effective way to work with raster engraving operations is to be able to set the min and max S value. This allows you to achieve very accurate contrast control on the material you are engraving.

The Beam Range is correctly located in settings as this sets the overall range your machine will expect S values.

The Beam power needs to have a min & max value in the operation.

For instance, if a user set:
Beam range to 0-1
Bean power 25-85%

then the resultant gcode would have S values ranging from 0.25-0.85.

jorgerobles, 2017-02-05:
@DarklyLabs  here it is!

DarklyLabs, 2017-02-05:
Changing the Max laser power value causes the Min value to change to 1/10 of the Max.
They should be independent values that do not affect each other.
Any combination should be possible, even Min > Max.

DarklyLabs, 2017-02-05:
It opens up the possibilities to create interesting effects with engravings.
It is allowed on rater-to-gcode.js.

jorgerobles, 2017-02-05:
easy peasy! Please check.

DarklyLabs, 2017-02-05:
@jorgerobles Perfecto! Thanks a million.

(by the way, that little bug with the setting tab not opening seems to be back today)

jorgerobles, 2017-02-05:
Ough I will check (zombie issues)

closed 2017-02-06 by jorgerobles