LaserWeb/LaserWeb4 #141: Chrome 'Page Unresponsive' message

DarklyLabs, 2017-02-04:
The 'Page Unresponsive' message regularly appears during long gcode generation operations.
Often pressing the 'wait' button results in incorrectly generated gcode.

Can this be overcome?

[image]

jorgerobles, 2017-02-04:
@lautr3k?

tbfleming, 2017-02-04:
Not @lautr3k's problem. Gcode generation needs to eventually move to a web worker.

lautr3k, 2017-02-09:
@DarklyLabs you can use the [lw.raster-to-gcode demo page](https://lautr3k.github.io/lw.raster-to-gcode/dist/example/) until the worker is implemented, the demo use a timer to avoid blocking the UI.
PS.: @tbfleming you can enable it with the `nonBlocking` setting ;)

jorgerobles, 2017-02-09:
@DarklyLabs non blocking implemented!

DarklyLabs, 2017-02-10:
@jorgerobles Just ran a few tests. Didn't get a non-responsive message at all.
The only thing we noticed is the need to keep an eye on the amount of ram Chrome is using during big rastering tasks. We managed to lock up OSX a few times by running our of ram.

DarklyLabs, 2017-02-10:
@jorgerobles By the way, the progress indicator is AWESOME!
Well done.

openhardwarecoza, 2017-02-10:
@DarklyLabs made me do a git pull to see what he is talking about... Had to make a gif, to confirm AWESOME indeed

[image]

closed 2017-02-16 by jorgerobles

cprezzi, 2017-02-17:
@jorgerobles Is it possible that generation of raster gcode is much faster since you added this feature?
It seems so to me!

jorgerobles, 2017-02-17:
I wish! :smile: