LaserWeb/LaserWeb4 #142: Where to place firmware specific gcode optimization?

cprezzi, 2017-02-04:
Different firmwares need different gcode optimizations to get best performance (or even just working code). Where should we place these optimizations?

It doesn't make sense to implement it into every gcode-generator in the cam. I also think that the cams should produce clean human readable gcode (not optimized).

The optimizer could be a class or lib that is only used before sending a job to the machine. For saving gcode, the user should have the option to select clean or optimized gcode. 

At the moment, I have some gcode optimizations in lw.comm-server, which probably is the wrong place.

cprezzi, 2017-02-04:
On the other side, some optimizations depend on the streming protocol and need to be implemented in the server.

jorgerobles, 2017-02-04:
Yes, I've been wondering about a postprocessor chain, but it has to be well
planned.  One of the things we should leverage first is a webworker
framework.
In the meantime lib/postprocessor could work enough

jorgerobles, 2017-02-16:
@cprezzi does this issue still valid. I mean, I agree with you the optimizations about hardware should be with the hardware handling code (comms-server)
Postprocessor chain would be a cool thing to add, but should be on other issue and is not vital by now.

cprezzi, 2017-02-16:
I think we could close this issue for now. If we implement a prostprocessor chain, it will be later (after the release).

closed 2017-02-16 by jorgerobles