LaserWeb/LaserWeb4 #143: Issues generating vcarve gcode

cojarbi, 2017-02-05:
I wanted to try some actual v carving but im getting this crazy gcode. Is it me ?
[image]
[image]
[Uploading gcode (29).gcode.txt…]()
[Uploading rtm-snapshot.json.txt…]()

jorgerobles, 2017-02-06:
@cojarbi Cannot download the sample files :|

cojarbi, 2017-02-06:
@jorgerobles fixed, now you should be able to see them

jorgerobles, 2017-02-08:
@cojarbi, indeed it goes nuts. I've tested each character and goes worse. Please attach the original SVG for stray points or so.

Bad news is VCarve is made with genuine Black Magic by our resident nigromant @tbfleming (https://github.com/LaserWeb/web-cam-cpp.git).

cojarbi, 2017-02-08:
Here you go
[RTM.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/760493/RTM.svg.txt)