LaserWeb/LaserWeb4 #145: npm start not opening correct webpage (Win10)

cprezzi, 2017-02-07:
When I start the dev environment with `npm start` a browser is open with url `0.0.0.0:8080` which is not working on Windows 10. Could we use localhost:8080 instead? Would that work on all supported platforms?

jorgerobles, 2017-02-07:
Yes, will work. It's located on package.json so easy to change.

cprezzi, 2017-02-07:
Actually it is in webpack.congif.js and I have changed it already on my system. I just was not sure if I should push it to git.

jorgerobles, 2017-02-07:
Yes, of course, thanks :)

jorgerobles, 2017-02-07:
Hmm, seems trickier than I thought. Localhost wont allow remote clients to access (same lan, etc)

openhardwarecoza, 2017-02-07:
Since the average user never has to deal with it, do we really need to autostart chrome?
The original issue, just asked that we print the IP/Port it listens on in the nodejs terminal like lw3 does

The thing is, unlike LW3, normal users wont run LW4's Webpack server at all...  We'll use webpack to pack a dist, which will pair up with lw.comms-server.  LW.comms-server can print its IP/port. 

So.... redundant?

jorgerobles, 2017-02-07:
Well, was requested on https://github.com/LaserWeb/LaserWeb4/issues/104
I tested changing to localhost on webpack.config.js to 
` devServer: {
        contentBase: dist_path,
        inline: false,
        hot: true,
        host: 'localhost'
    },`

And seems working well enough ...but I cannot foresee any issue arising changing this.

openhardwarecoza, 2017-02-07:
(: if its fixed its better

jorgerobles, 2017-02-07:
@cprezzi Is it working for you with this patch?

cprezzi, 2017-02-07:
It's working for me with only changing `devServer: { .... host: 'localhost' }`

jorgerobles, 2017-02-07:
Perfect :)

closed 2017-02-07 by jorgerobles

tbfleming, 2017-02-07:
0.0.0.0 works on my Windows 10 machines and my Linux VMs. It means bind to all addresses; it allows remote connections unlike localhost. The confusing part: 0.0.0.0 is valid for servers, not clients. Chrome has to connect to localhost or the appropriate IP address if remote.