LaserWeb/LaserWeb4 #146: Failure to render all SVG

maudette01432, 2017-02-08:
Having an issue with LW4 rendering SVG.   I have provided a simple test case out of inkscape with the letters TUIT as closed paths as written out of inkscape.  The cam operations can see the paths but they are not rendered in the viewer.  One letter is red the others are black.  

See screen shot below (prior to changing fill of first "T" to red.

[image]

Source file having issues with.  Because it's so simple hopefully it can be used to resolve the problem.  The "T"s in question have only 32 nodes.

[tuit.svg.zip](https://github.com/LaserWeb/LaserWeb4/files/761945/tuit.svg.zip)

This renders improperly on both Mac OS X Sierra w/ chrome and Windows 10 w/ chrome.

maudette01432, 2017-02-08:
Interestingly the operations seem to work even though it doesn't render like this laser_fill_path operation which does a crosshatch.

[image]

The vcarve operation...while it seems to engave the letters OK seems to also have some big vertical lines rendered.

[image]

The GCODE on this is definitely bad.  Path 18 seems to be off in la-la land.

jorgerobles, 2017-02-08:
@lautr3k got this on console loading up the file attached. Could be of help?

[image]

lautr3k, 2017-02-09:
Work great on the lw.svg-parser demo page (https://lautr3k.github.io/lw.svg-parser/dist/example/) :
[image]

jorgerobles, 2017-02-09:
@lautr3k, @tbfleming  After a while I've tracked down the problem to LW4 gets choked with relative coordinates. "m" coordinates are not rendered properly. 
AI converts to "M" coordinates and run ok, Inkscape use "m" ones and shows nothing, I think it goes outside GL object bounds.

[tuit_original_AI.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/763607/tuit_original_AI.svg.txt)

tbfleming, 2017-02-10:
Laser Fill Paths won't work if there's a problem with lw.svg-parser. The bugs happen after that stage.

Bug 1:  some shapes hit a limitation in the poly2tri library. This only affects the canvas. We need a replacement, but every triangulation library I've tried (even three's) hits issues, they just vary with which conditions cause them to fail.

Bug 2: some shapes cause vcarve to mess up badly. It may be a while before I get around to looking at vcarve again.

tbfleming, 2017-02-12:
Bug 1: I added a workaround until someone gets a round TUIT for making a decent triangulation library. It now shows a red stroke if poly2tri fails and the original stroke is invisible.

Bug 2: track in #143 

[image]

openhardwarecoza, 2017-02-28:
Working well enough, no recurrence of complaints. Closing

closed 2017-02-28 by openhardwarecoza

maudette01432, 2017-02-28:
I haven't complained because I thought it was still being fixed?!

For the record the work around is nice...but I still can't see the fill colors or the outline colors for selecting operations.

jorgerobles, 2017-05-22:
A light review of this issue, the original file throws colinear error on latests poly2tri. 
So I guess that T where not closed paths. Opening on AI autofix colinear.
[image]

jorgerobles, 2017-05-22:
@tbfleming https://sourceforge.net/p/jsclipper/wiki/Home%206/ B4.2 Cleaning, can we apply this clean method to polygons whenever that exception arises?

tbfleming, 2017-05-22:
I already use Clipper to clean it. Colinear doesn't mean the path isn't closed. The JS port of poly2tri doesn't cover the hardest cases that the original C++ version does. These harder cases are all valid polygons that Clipper, boost::polygon, and the rest of our code is OK with. Last time I looked, none of the JS triangulation libraries handled all hard cases, but different ones handle different subsets of the hard cases.