LaserWeb/LaserWeb4 #147: Rotate vectors / images

DarklyLabs, 2017-02-10:


jorgerobles, 2017-02-24:
@tbfleming Is this very difficult to implement?
I've tried https://webglfundamentals.org/webgl/lessons/webgl-2d-rotation.html but seems scaled on rotation. Maybe relates with 2d transforms on 3d viewport.

tbfleming, 2017-02-24:
@jorgerobles scale and rotate get along with each other as separate fields, but the math gets complex when you add a rotation field since all 3 fields interact with each other. The easy method is to switch to a single matrix:

* Drop document's translate and scale fields and replace them with a matrix. The single matrix will handle translate, scale, rotation, and skew. gl-matrix has functions to help with this.
* Rewrite code in document-cache and workspace which deals with document bounds; the approach I currently have uses an optimization which is incompatible with rotation.
* Change the vertex shaders to use a matrix uniform instead of translate and scale uniforms
* Update the code in the workspace which manipulates translate and scale to manipulate the matrix instead
* Change ```rawPathsToClipperPaths``` to use the matrix instead of separate translate and scale fields. This will make all of the cam's vector operations work correctly.
* Add rotation to lw.raster-to-gcode; it'd be nice if we have a second person who works on this code.
* Add code to update the documents in old files.

jorgerobles, 2017-02-24:
Thanks!.. Seems daunting 😨

openhardwarecoza, 2017-02-24:
(: theres always the "LW isnt a CAD package" argument (;  
Simple positioning yes, rotating might count as "rather use cad"

DarklyLabs, 2017-02-26:
Move, Rotate and Scale are basic manipulations needed for placement. Even the ability to just rotate a design 90degrees CW or CCW would be beneficial.

Users shouldn't have to jump back into a CAD package if they simply wanted to rotate a design to fit   a free area of material they have.

tbfleming, 2017-02-26:
90 degrees simplifies things. For vectors, edit the rawPaths; you don't need to touch the shaders. For bitmaps, run them through a transform then store them back into dataUrl.

tbfleming, 2017-02-26:
You'll also have to adjust scale and translate to compensate.

jorgerobles, 2017-06-01:
Powerful image manipulation https://www.npmjs.com/package/jimp

domenic-d, 2017-06-01:
@jorgerobles Will this handle vectors as well?

jorgerobles, 2017-06-01:
Seems not. I just posted as a reference.

tbfleming, 2017-06-01:
@domenic-d I'll take care of vectors. @jorgerobles is looking for a way to handle images without relying on separate translate, scale, rotate fields, which will go away. I'm going to replace them with a matrix.

tbfleming, 2017-06-01:
Here's why we have to shift to matrices for rotation. I said above "the math gets complex when you add a rotation field since all 3 fields interact with each other." I sugar coated that. In some cases the math becomes impossible to solve. Here's one of those cases.

[image]

If we have separate rotate and scale fields, then we have to decide which one happens first. If we scale the original object then rotate it, then A can transform into B, but B can't transform into A. If we rotate first, then B can transform into A, but A can't transform into B. Either approach adds a restriction to the UI. e.g. if we rotate first, then a user wouldn't be able to rotate an object they already scaled.

Matrices doesn't restrict order. The user can freely rotate and scale in any order. A major downside is lw.raster-to-gcode doesn't understand it. @jorgerobles will have to either preprocess the image before lw.raster-to-gcode or do serious surgery on lw.raster-to-gcode.

jorgerobles, 2017-06-01:
I will go preprocessing, even we have to store the original and the transformed one, will be better.
I would like @lautr3k could lend us a hand... are you there? 😇

jorgerobles, 2017-06-01:
@tbfleming as we talked before please do a branch, I hope not, but can be a long task.

tbfleming, 2017-06-01:
I just pushed a new branch: rotate. It can now rotate vectors. I haven't ported the workspace handling of images to the matrices yet. Image display is currently broken.

todo (me):
* Fix workspace image support (loading, loading in SVG, displaying, selecting)
* Fix dxf import
* Translate old workspace files to new format

domenic-d, 2017-06-01:
@tbfleming Thanks for the explanation. Please let me know when you need some specific testing performed and i will jump onto it.

tbfleming, 2017-06-01:
@domenic-d go ahead and test what's there. Don't do the following yet:
* Load an image
* Load an SVG that contains any images
* Load a DXF
* Load an old workspace file

domenic-d, 2017-06-02:
Its great to see this functionality making its way into LW.
Some feedback:
1: Typing rotation values works well.
2: Clicking the increase/decrease arrows sometimes over-ran the number of clicks made. Couldn't work out how to consistently replicate this.
3: Deselecting the document and then re-selecting it set its rotation back to zero. It would be useful to retain its rotation value so it can be reset to the original orientation.
4: The value window constantly moving as rotation was being applied made it a little tricky, especially when the increase/decrease arrows were being used.

tbfleming, 2017-06-02:
3 is tricky; there is no rotation value, so it creates a phantom one when the selection changes. When the field changes it computes a rotation matrix from the difference of the two values then multiplies that matrix to the individual matrix of each selected object.

tbfleming, 2017-06-02:
Ditto with size; there never was a size member; it always computed it on the fly.

tbfleming, 2017-06-02:
No. That creates the math problem I mentioned above.

domenic-d, 2017-06-02:
Thanks Todd. Having this feature in LW is a huge benefit. If the user has to eyeball the rotation or even reload the doc then so be it.

tbfleming, 2017-06-03:
It can now load image files and rotate them.

tbfleming, 2017-06-03:
@jorgerobles: here's how the doc is now set up: The image occupies a rect with bottom-left corner at 0,0 and upper-right at W,H. The conversion to WCS assumes 1 pixel in the bitmap is 1mm. So, a 1000 x 1000 pixel bitmap is 1m x 1m. The document's transform2d matrix scales, rotates, and translates it to the final WCS bounds.

tbfleming, 2017-06-03:
Users may combine Rotate & Scale to skew an image:

[image]

cojarbi, 2017-06-03:
Work as expected with the same points described above
clicking the restore icon ( back red arrow ) makes the image disappear

cojarbi, 2017-06-03:
@jorgerobles restore is working now. Will it be able to restore also the rotation? Correctly it does not

tbfleming, 2017-06-03:
It now loads transformed images embedded in SVG files.

tbfleming, 2017-06-03:
Bug note: you may have to ungroup everything in Inkscape if you do transforms because of #332.

tbfleming, 2017-06-03:
Group bug fixed.

tbfleming, 2017-06-03:
@domenic-d 

> 5: Size values change as the doc is rotated. I am in two minds about this. On one hand, it would be good to not change the size (since rotation shouldn't be changing it), but the way it is expressed (as x&y) makes sense to have it show a change.

I just tried this on Inkscape. It does the same thing we do with the size. It also doesn't have a rotation value that the user can reset that I could see.

tbfleming, 2017-06-03:
DXF updated. Not tested.

tbfleming, 2017-06-04:
It can now load old workspaces.

domenic-d, 2017-06-04:
@tbfleming  Just back to testing now.
The sizing with rotation is fine the way it is. Users can just make sure they keep the x-y size locked and they won't experience any skewing of their image.

What is the intention of the red undo arrow with raster images? It currently distorts the image when applied.

domenic-d, 2017-06-04:
DXF loading and rotating  looks to be working as expected.
Image loading and rotating looks to be working as expected.

Raster Opt & Reset distort image when a rotation other than 90 degrees has been applied.

jorgerobles, 2017-06-04:
Red undo is related with raster optimization, that adjusts dpi to laser
beam, many months ago #132.

Last patch commited only works with scale. But has to be fixed as now scale
and rotation goes together.

jorgerobles, 2017-06-04:
I'm almost completed the first approach to raster transforms.

Pending bugs
- image offsets
- A axis deformation

[image]

[image]

tbfleming, 2017-06-04:
@jorgerobles: I added this after context creation to expose what's happening:

```
ctx.fillStyle = 'black';
ctx.fillRect(0,0,w,h);
```

[image]

It's aligning the lower-left corner of raster-to-gcode's output to one of the transformed corners of the image. You need to align center-to-center instead.

jorgerobles, 2017-06-04:
Ugg can't get it right. Tried translation on transform2[4] and [5] but nup.

tbfleming, 2017-06-04:
@jorgerobles I fixed that part for you.

jorgerobles, 2017-06-06:
@tbfleming I've matched Raster A Axis transform to dev-es6 branch behaviour, maybe vector offsetting is not?

[image]

tbfleming, 2017-06-06:
@jorgerobles fixed.

tbfleming, 2017-06-06:
Oops. That introduced a new bug: laser spacing will be off.

To fix that, don't use axisAFactor in params. Instead scale the output gcode.

jorgerobles, 2017-06-06:
Reducing Y ppi, or postprocessing?

tbfleming, 2017-06-06:
postprocessing

tbfleming, 2017-06-06:
Found a different bug while testing. I used the del key shortcut on vector objects. That left the children members of the parents untouched, which causes all kinds of issues.

jorgerobles, 2017-06-06:
Ok, will dig on that

jorgerobles, 2017-06-06:
@tbfleming, multiply beam diameter by aAxisFactor in params is not enough?

tbfleming, 2017-06-06:
@jorgerobles beam diameter affects both X & Y spacing.

domenic-d, 2017-06-07:
Just running some more testing on the rotate. Seems to be working correctly with both vectors and raster images.
Is it at a point it could be merged in with dev-es6?

tbfleming, 2017-06-07:
It looks like it. @jorgerobles any more issues you know of?

jorgerobles, 2017-06-08:
Nup. I think is ready ;)

domenic-d, 2017-08-11:
If no one has any objections, I believe this can be now closed?

closed 2017-08-11 by jorgerobles