LaserWeb/LaserWeb4 #148: Feature request - touch plate for Z height AND edge finding

funinthefalls, 2017-02-11:
Would it be possible to add a feature like this?

https://www.youtube.com/watch?v=bhZdbgM6S70

I especially like that it compensates for work pieces at an angle, not perfectly parallel to the X and Y axis. I am willing to help with a #beerfund, #palletfund, or #redbullfund donation to get this rolling :)

openhardwarecoza, 2017-02-11:
Make a macro button.  Existing probe+set zero gcodes...

funinthefalls, 2017-02-11:
I understand how to probe and set the Z axis, but I am at a loss to figure out how to do the edge finding for x and y. Any help would be greatly appreciated.

openhardwarecoza, 2017-02-11:
In the exsiting dro we already have all that built in.  In lw3,  dro,  dropdowns next to each axis dro

closed 2017-02-11 by openhardwarecoza