LaserWeb/LaserWeb4 #149: 4th Axis TO THE MAX!

FabCreator, 2017-02-11:
So far not much has been done with the 4th axis, with our kits starting to leave us really soon, i am hoping i can turn my attention to continuing to develop the hardware :)

So i thought id through a few ideas I have for the 4th axis:

I was thinking about engraving cylinders which are not straight, like glasses bottles etc they have shape to them.

Would it be possible to create a Z path for the laser to follow. (example below)
[image]

This way you could easily photograph the object side on sketch the curve and then you have the ability to engrave non standard objects without the need for a fancy sensor.

Another 4th axis use is a motorised pass through, engrave the board you have and then the 4th axis drive the board to the next area. this could be used as part as an auto feeder ;)

Cheers,
Bonne

DarklyLabs, 2017-02-15:
This is a great idea.

openhardwarecoza, 2017-02-28:
@tbfleming something you'd consider or something we'll close off?  (:

tbfleming, 2017-02-28:
Not anytime soon

closed 2017-02-28 by openhardwarecoza