LaserWeb/LaserWeb4 #152: Google Chrome timeout and progress

opaque01, 2017-02-12:
I tried to generate GCODE and Chrome won´t react. If I don´t open activity view, I don´t know if chrome (LaserWeb4) is still working.

Is it possible to change the timout message (wait - close)?
Is it possible to show a progress bar?

greetings opaque

[image]
[image]
[image]
[image]

tbfleming, 2017-02-13:
Duplicate of #141, which is implemented for bitmaps, but not yet for vectors

tbfleming, 2017-02-17:
Reopening since #141 was closed

jorgerobles, 2017-02-17:
@opaque01, please attach source file and workspace json
@tbfleming what would be the best way to "progressize" the vector operations?

tbfleming, 2017-02-17:
Create a web worker to call getLaserCutGcodeFromOp() and getMillGcodeFromOp(). You'll have to create a way for the web worker to communicate with the main thread.

jorgerobles, 2017-02-18:
@tbfleming I've done tests, and doubt webworker will do anything. It makes the processor run on his own thread, but the thread could stall equally on large computations. So I'm going for the setTimeout 0 hack, but need to know where to "slice" the code to run.

I've seen a lot of loops and whiles in cam-gcode-mill. As I don't fully understand all the operations done there, please tell me the lines ideal to be splitted for recurrent function calls.

tbfleming, 2017-02-18:
Part of the code is C++ compiled to javascript; the setTimeout 0 hack won't work for those parts. e.g. the entire v-carve algorithm is in C++. The setTimeout hack leads to confusing code, especially for complicated algorithms like the CAM has. It's best to avoid it.

There's an advantage to webworkers: long runs don't hurt the UI thread. A cancel button can kill a webworker (worker.terminate()) if the user decides it's been running too long. Also, Chrome normally doesn't complain about busy webworkers.

jorgerobles, 2017-02-18:
@tbfleming Current issue will not be solved if we have a 100% process hogging the cpu. Also, yesterday I let a simple carve (not vcarve) with an overstep of 0.1 for 2 hours and was still hung. No UI interference (I left the computer alone and sad, an busy)
Also, if hung, no progress will be sent to the main process so apart the moving UI, the gcode will not be generated (though be cancelled)

Any ugly code could be refactored when manageable portions get achieved. I like to do things well, but better things working.
Please, if you could tell me where will be the best place to start slicing the code, at cam-gcode-mill.js:77, 90 or 112.... or other

jorgerobles, 2017-02-18:
@lautr3k, could you implement an stop method to lw.raster-to-gcode? thanks

tbfleming, 2017-02-18:
> Current issue will not be solved if we have a 100% process hogging the cpu

It will solve it:
* The UI won't freeze
* Chrome won't open a terminate-or-wait dialog
* The user can cancel without having to kill and reload the page

> Any ugly code could be refactored when manageable portions get achieved

The SetTimeout 0 hack turns code inside-out, making it hard to understand. The CAM code already takes a lot of effort to understand. Applying SetTimeout 0 to it will make it much harder.

jorgerobles, 2017-02-24:
@lautr3k working on this, seems Webworkers cannot operate DOM :| . lw-rastertogcode cannot be converted to use webworkers unless refactor and accept a Binary bitball of the image.

jorgerobles, 2017-02-24:
@tbfleming if have to move  getLaserCutGcodeFromOp() and getMillGcodeFromOp() to webworker, they have to show send a progress message (in any of the former lines)

tbfleming, 2017-02-24:
@jorgerobles they can use postMessage()

jorgerobles, 2017-02-24:
Yes.. on which line is better? (77...) 😏

tbfleming, 2017-02-24:
Let's focus on getting it working first. 0% at beginning and 100% at end.

jorgerobles, 2017-02-24:
@tbfleming some CAM operations or dependencies (snap?) rely on DOM, window, or regular process. 
Cannot be used on webworkers. https://github.com/LaserWeb/LaserWeb4/tree/webworkers

tbfleming, 2017-02-25:
@jorgerobles I pushed a fix to the webworkers branch.

jorgerobles, 2017-02-25:
Webworking! (pun intended at https://github.com/LaserWeb/LaserWeb4/commit/4f5040243ca7b83aad3755c6549e183d7f5dfa66)

jorgerobles, 2017-02-25:
@tbfleming examineDocTree moved to webworker.

tbfleming, 2017-02-25:
@jorgerobles examineDocTree works with images; it can't move to a webworker.

lautr3k, 2017-02-26:
My first commit used webworker and @tbfleming asked me to remove this feature...

jorgerobles, 2017-02-26:
@tbfleming I've ported that to webworker late night. It works well. I will merge branch today if lw-raster2gcode PR is solved. 

Checkout https://github.com/LaserWeb/LaserWeb4/tree/webworkers for testing before commit.

@lautr3k I've done a pull request to lw-rasterizer, implementing terminate to emulate a webworker, please check before commit. Now I'm using my own fork, I prefer yours :smile:

@lautr3k Also, the webworking issue of raster is the manipulation of Image done in lw-canvasgrid. If lw-canvas-grid have a method to convert to bitball on the main thread and then use the bitball on webworker thread instead "image" will work.

tbfleming, 2017-02-26:
The CAM didn't have a job management queue yet, so couldn't handle it. If I remember right, you kept it as an option.

jorgerobles, 2017-02-26:
@tbfleming I was meant for me? I did not understand. There's a job queue now for handling webworkers.

tbfleming, 2017-02-26:
Sorry, I meant @lautr3k

jorgerobles, 2017-02-26:
Merged and live! Will improve on @lautr3k works about webworkers.

jorgerobles, 2017-02-28:
@opaque01 should be working now as expected.

closed 2017-02-28 by jorgerobles