LaserWeb/LaserWeb4 #155: Let's steal from the Chinese (;

openhardwarecoza, 2017-02-15:
So, in the news today, Eleksmaker, seems to have pulled some inspiration from LW3 

``` 
if ( "Imitation is the sincerest form of flattery" == true ) {
    console.log("LW is so famous the Chinese even copy us")
}
```

[image]

On a sidenote, EleksCAM seems to be more of a "skin" over Benbox than an outright copy.   At least it doesnt appear to be nodejs. 

[image]

That said, they do have one or two cool features we might want to steal from (fair game right?)

#  View quick set
[image]

# Gallery of included jobs
[image]

FabCreator, 2017-02-15:
Both of those a very nice features. Also I belive they say opensource. So it's fair game. Let me find a documentary for you Peter. Its about the china and how everything is opensource they just don't share. ;)

FabCreator, 2017-02-15:
https://www.youtube.com/watch?v=SGJ5cZnoodY

jhiner1978, 2017-02-16:
Steal away, I say. I would love to have a gallery of previous jobs so I can go back and engrave/CNC old jobs. As for 3d view select, that is a great idea as well. Would be even better to add arrows under each view to rotate said view. Corner views would be cool too like fusion 360.

cprezzi, 2017-02-16:
I belief the 3d view buttons don't make sense at the moment. All supported operation are max. 2.5d, so the only useful views are from top or diagonal. From the sides it would only be some lines in differen depth.

cprezzi, 2017-02-16:
But I like the History feature. This could list the history of all generated cam operations, which could be recalled (including all cam settings).

openhardwarecoza, 2017-02-16:
Ahh you just took Gallery as I was thinking of it (ie some sample DXFs) and
added the idea of Past Jobs! (Localstorage isnt big enough, but hey we
already have workspace export, all we need is a way to dump a preview
render and store it in nodejs backend)

openhardwarecoza, 2017-02-16:
Agreed on the side view, but i was thinking actually instead of the current
"Reset view button" - if you move it around with the mouse and want to
returns to some view.  Maybe not side, sure, but maybe standards like
isometric, corner perspective, top down, etc?

cprezzi, 2017-02-16:
I was allready thinking about a job history store on the server and also sending back the running gcode in case the browser was closed an reopened (or a second browser is connected), but a CAM history would be even better, because you could change some settings before re-run the job.

openhardwarecoza, 2017-02-16:
I love that idea!

closed 2017-07-08 by openhardwarecoza