LaserWeb/LaserWeb4 #156: Add section to view / inspect gcode text

cojarbi, 2017-02-15:
We have a section for this in LW3

tbfleming, 2017-02-15:
We had it for a while in LW4; it killed performance. Chrome has issues with fields containing huge text blocks (Firefox is much worse). React magnifies that.

gndprx, 2017-02-16:
If that is the component that cripples my browser in LW3, then I'm all for leaving it out.

If anything, maybe a button to pop-up the generated g-code for review for anyone who really wants to review it in session.

openhardwarecoza, 2017-02-16:
there is a download button on the cam tab (familiar stiffy save icon)

cojarbi, 2017-02-16:
The download is there, it just happened that I needed to see the code before executing but not at the expense of usability or speed. If there is no way to add a preview we can close this

jorgerobles, 2017-02-19:
@cojarbi take a look for a shiny view gcode button :)

cojarbi, 2017-02-19:
Very elegant as always @jorgerobles thanks

closed 2017-02-19 by jorgerobles