LaserWeb/LaserWeb4 #158: Add Google Analytics

openhardwarecoza, 2017-02-16:
```
<script>
  (function(i,s,o,g,r,a,m){i['GoogleAnalyticsObject']=r;i[r]=i[r]||function(){
  (i[r].q=i[r].q||[]).push(arguments)},i[r].l=1*new Date();a=s.createElement(o),
  m=s.getElementsByTagName(o)[0];a.async=1;a.src=g;m.parentNode.insertBefore(a,m)
  })(window,document,'script','https://www.google-analytics.com/analytics.js','ga');

  ga('create', 'UA-38524928-2', 'auto');
  ga('send', 'pageview');

</script>
```

Ps: if you want admin access to the Analytics profile just PM me your email address 

In LW1-3 I used analytics just to track user counts and basic demographics.  Then i share that with the g+ community around the 25th of each month 

If any of you are bigger Analytics wiz'es i wonder if we could extend with some custom properties like Firmware in use,  job lengths,  or other cool stats that would be useful to plan for the products future?

jorgerobles, 2017-02-16:
Done!

openhardwarecoza, 2017-02-16:
thanks @jorgerobles 

(; keeping it open though so we can investigate "If any of you are bigger Analytics wiz'es i wonder if we could extend with some custom properties like Firmware in use, job lengths, or other cool stats that would be useful to plan for the products future?"

openhardwarecoza, 2017-02-28:
Closing for now

closed 2017-02-28 by openhardwarecoza