LaserWeb/LaserWeb4 #160: Order or operations not honoured.

DarklyLabs, 2017-02-19:
The order of operations are not being honoured when gcode is generated.

Step 1:
Add 3 documents.
Lets call them: 
Doc1
Doc2
Doc3
Note, this is how they appear in the document list.

Step2:
Drag Doc2 into an operation
Then Doc1
Finally Doc3

Step 3:
Generate GCode.

The order of the gcode operations are Doc1, Doc2, Doc3 instead of Doc2, Doc1, Doc3.

jorgerobles, 2017-02-19:
@DarklyLabs Hmm "Order of documents in operation" is clearlier :smile:
Actually there's no way to order inner documents in operation (not a bug, just unimplemented), but a workaround is creating single operation for each document.
[workspace_#160.json.txt](https://github.com/LaserWeb/LaserWeb4/files/785684/workspace_.160.json.txt)

DarklyLabs, 2017-02-19:
It seems more natural for the operations to happen in the order required by the user. For instance if a user wants to first perform an engraving task and then cut out the design so the workpiece does not move as would happen if the cutting task was performed first.

The document order should not be relevant. Many documents may be loaded and used in various ways. Documents may even be loaded and unused.

For your work around to work there needs to be a way to organise documents, since this seems the order to be used for gcode generation.

DarklyLabs, 2017-02-19:
@jorgerobles Sorry if my initial explanation was not clear.  I am talking about separate operations created for each document loaded. 
The order of gcode generated is based on the order of the documents are loaded instead of how they are used to create operations.

jorgerobles, 2017-02-19:
I'm a bit lost :)

@DarklyLabs please download workspace file attached, rename to .json and load on LW using CAM:workspace:load button on top.
Generate gcode. The operation order is honored. You could jumble using the UP/DN buttons each operation below and regenerate. 

[workspace_#161b.json.txt](https://github.com/LaserWeb/LaserWeb4/files/785710/workspace_.161b.json.txt)

DarklyLabs, 2017-02-19:
@jorgerobles Your example works correctly.
Lets consider this closed and I will advise if I can repeat the situation consistently again.
Thank you.

closed 2017-02-19 by DarklyLabs