LaserWeb/LaserWeb4 #162: Put Console back to bottom right corner?

cprezzi, 2017-02-19:
Could we please place the console back to the bottom right corner (like on LW3)?
The messages in the console are important for machinig, but are often out of view on many screens. It also doesn't make sense to have the console on multiple tabs (com & jog).

[image]

openhardwarecoza, 2017-02-19:
I agree (:

tbfleming, 2017-02-19:
Does the console serve a purpose outside of comms and jog?

openhardwarecoza, 2017-02-19:
In lw1-3 it served as all round notifications (updates available,  news,
 progress updates,  syslog,  error log,  etc)

tbfleming, 2017-02-19:
And is promptly ignored by most users because of this large amount of messages, even me, except when I'm entering gcode.

openhardwarecoza, 2017-02-19:
I wouldnt say "most" ...

Most of the old hands (and very quickly the new guys realise it too) know
the syslog contains valuable info when things go wrong.  It also helps us
to say "check the syslog, was there any errors about xyz"

In that argument though, position wouldnt be the problem, the information
being displayed would be the issue (adjustable verbosity might be a better
idea - ERROR/WARN/INFO/DEBUG?)

@lautr3k 's Optimisation rewrite of LW3 (which sadly never got used)
actually had a syslog that majorly improved on what was in lw1-3 :  See
https://github.com/LaserWeb/lw.comm-server/issues/2#issuecomment-274247359

cprezzi, 2017-02-20:
@tbfleming Yes, I think the console can be of use on all tabs for notification. Often I switch to other tabs while running a job and then I would like to still see the notifications. 

Also, the spece @ bottom right is feeling soo lonely empty ;)

cprezzi, 2017-02-20:
Yes, I think the console could be of value on all tabs for notifications. I often change to other tabs while running a job but don't want to miss a notificaion. The actual position in the com/jog tab is often out of view and therefore useless.

But the main reason to move it there is, the space at bottom right is soo empty lonely ;)

cprezzi, 2017-02-20:
We could implement a verbose level to reduce the reports in the console, like I have allready done on server side.

jorgerobles, 2017-02-20:
I could put back in a minute, just say the word

openhardwarecoza, 2017-02-20:
"the word"  lol

jorgerobles, 2017-02-20:
B-bird's the word oh well-a bird bird bird

closed 2017-02-20 by jorgerobles

cprezzi, 2017-02-20:
This was quick, probably a bit too quick. ;)
The console/syslog should be right of the existing elements, not bellow. This way it takes too much space.

jorgerobles, 2017-02-20:
There ;D

cprezzi, 2017-02-20:
Yep, right there :) Thanks!

commit ab7094e4 (2017-02-20): Fix #162
commit f550f4b8 (2017-02-20): Fix #162