LaserWeb/LaserWeb4 #165: Raster problems

haqnmaq, 2017-02-22:
This is a problem that I have noticed for a little while with Laserweb 4. I originally posted the problem on G+ https://plus.google.com/u/0/101842495658501740623/posts/D28hrdj3vrU  but I have since done more tests and I have gotten similar results. 

[image]

The problem is only with the new raster engine in LW4, and the exact same files raster fine in LW3 with the same power and speed settings. There seems to be alot more jerkyness when it rasters, and the laser fires at weird times, almost exactly where it shouldn't. For instance it will fire the laser will fire where it should be white and turn off when it should be black, but on the next pass (next pass up one layer) it will fire correctly according to the image. 

I have also noticed that even though the gcode preview in Laserweb3 and 4 shows that the laser will travel from one side all the way to the other, when the laser actually etches, it etches about half way, then back, then on the next pass it will go the full X distance and back, then on the next maybe only go 30% across and back. It seems very glitchy, and not what a normal raster would do. 

I did some more tests earlier tonight. Here are some images of the results.

[image]

On the left is a smaller version of the test before, but as a .png file instead of the .jpeg as the first test. The file type didn't seem to make a difference. On the right, I tried a color .png image, and as you can tell it performed much better than the black and white images. I will note however that near the hair at the top it had a very similar issue as I described above. A couple of the passes didn't travel the whole distance, or it didn't fire the laser at the proper moment. (maybe this is only a problem with high contrast?)

[image]

I then tried another image. Again, this is the exact same file used on both versions of Laserweb. This was a black and white jpeg image. I tried it in LW4 and LW3 with the same speed and power settings. As you can tell, I had the same problem as with the other files. I also tried with the diagonal setting enabled and got really weird results. It would do the same thing as before (go part way and back, and fire at the wrong times) but it would also randomly move in the Y direction and X direction, and even raster in those directions as well. I captured the strange movements of the Diagonal Engraving on video that you an watch here. https://drive.google.com/open?id=0B8ZXvH-aGzn9TkNVZDZRQjBacTQ  

I don't have video of the other ones, but the laser basically behaves the same as in the video, but it only has that 'glitchyness" in the x axis instead of both axis in the video.

I have uploaded all my files from the gcode output, to the source images, and even some screenshots while I was running these tests here. https://drive.google.com/open?id=0B8ZXvH-aGzn9UXg0QmRQdVRyc1E  If you have any other questions, then let me know!

I used Laserweb 3 version 30552 and the Laserweb 4 version that was today 2/21/17 (not sure what version that would be) on a K40 running smoothie for all of these tests except for the initial one that was posted in the G+ community. I would have been running an earlier version for them.

jorgerobles, 2017-02-22:
Possibly related with #99
Any other thoughts @lautr3k?

tbfleming, 2017-02-22:
This appears to be the lone S bug. It confuses Smoothieware since a line with just an S# is invalid.

openhardwarecoza, 2017-02-22:
I dont know if @lautr3k disappeared or is angry at us,  but its been weeks
since he replied.  We may need to go dig into the code ourselves :(

tbfleming, 2017-02-22:
@haqnmaq Try running this gcode through; I stripped out the bad S# lines.

[FryBwLW4Bad fixed.gcode.txt](https://github.com/LaserWeb/LaserWeb4/files/792655/FryBwLW4Bad.fixed.gcode.txt)

tbfleming, 2017-02-22:
@openhardwarecoza If one of us fixes it then we'll have to abandon pulling it in via npm.

jorgerobles, 2017-02-22:
Or commit to a pull request to his repository or using a git branch on npm

openhardwarecoza, 2017-02-22:
Thats part of why i asked him if we can have that work in the organisation
(which he refused,  i suspect because of external poisoning).  The more i
work in Open source software,  the less happy i become.  Is that normal?
Lol.

In the organisation anyone could just push,  republish to npm,  etc.  He
would still have his credit,  he would still be the hero.  But
administratively we would all be better off.  It makes me extra nervous
because with whatever is going on which leads to the not wanting to work in
the organisation,  might end up with the modules disappearing from npm/git
anyway.  I keep forks,  but its extra work to keep forks up to date just in
case.  Sad situation actually.  Select individuals who want to see LW fail

jorgerobles, 2017-02-22:
Yes @openhardwarecoza,  but refocus. He made the code for laserweb as
stated in readme and is mit licensed.
Keep faith. We are free to fork, and contribute with Sebastien code. We use
a lot of code from others outside organization, and no kittens die.

When he will ready we will amend, accept our pull request and all fine.

Other way to work is what @cprezzi said. Make a postprocessor based on
hardware (a patch for what should be better patch on smoothie, yes)

Cheer up!

lautr3k, 2017-02-22:
> This appears to be the lone S bug. It confuses Smoothieware since a line with just an S# is invalid.
This is possible, but the "verbose g-code" option should correct this "problem" specific to the smoothie firmware. A single S value is valid g-code and I do not want to add specific code for each firmware.

openhardwarecoza, 2017-02-22:
Yay!  He's back!

tbfleming, 2017-02-22:
Lone S also causes grbl-love to slow down.

Doesn't Smoothie choke with verbose gcode?

tbfleming, 2017-02-22:
Autocorrect: grbl-lpc

haqnmaq, 2017-02-23:
@tbfleming Thanks, I will try that code tomorrow night (the laser cutter is at our makerspace) and get back to you. @lautr3k Thanks, I will also try the verbose gcode option and post the results.

haqnmaq, 2017-02-24:
[image] @tbfleming From the photo attached it doesn't  seem that the fixed gcode helped. It looks like it may have kept the lines, but in different areas. @lautr3k The "verbose gcode" did seem to work great! You can see the results in the photo. Thank you very much! I guess I should play with those raster settings more. At least until there is some documentation on what they do. Thanks again!

openhardwarecoza, 2017-02-24:
Theres a tonne of documentation on all the settings on
http://cncpro.co/index.php/44-documentation/operations-generate-gcode/operation-laser/laser-raster

#rtfm

tbfleming, 2017-02-24:
Since it's not the lone-S issue I'm starting to suspect a Smoothie bug.

cprezzi, 2017-02-24:
I can confirm that this issue does not happen with Grbl. I made tests with different params (M3, M4, trim pixels y/n, join pixels y/n, burn white y/n), but nothing produced this issue.

haqnmaq, 2017-02-28:
@openhardwarecoza Wow! There is alot more documentation there. I am sorry that I missed that. Thank you! I don't know who is doing the documentation, but it may help to put a line for the "verbose gcode" setting that if you are getting weird raster engravings, then try this setting. Or something like that. Feel free to use my photos to illustrate the "problem". 

I still can't figure out why it seemed to only do it in high contrast areas, but the verbose gcode setting solbed it, so it doesn't seem to be Laserweb related.

Thank you again for your help everyone! Feel free to close this issue out since it does seem to be a problem with smoothie and not laserweb.

closed 2017-02-28 by openhardwarecoza