LaserWeb/LaserWeb4 #167: Active Gcode "Generate" button

donkjr, 2017-02-23:
This may not be viewed as a bug. 
While learning LW4 there have been multiple instances where I could not get the image on the palate to behave as I expected. 
One tell tale indication is that the "Simulator" button would not do anything or after a change in setting it acted like nothing had changed. 
It seems that after an operation that changes the documents shade or the contents therein the GCODE menu: "Generate" button needs to be pushed.
This makes sense but it was not always obvious to me that I had done something that required the Gcode to be re-"Generated".
"As a LW4+ user I would like the "generate button" to actively indicate when something is changed in the settings or content that would expect the Gcode to be re-generated".

jorgerobles, 2017-02-23:
So any change on Settings, Operations, and Documents should mark a "dirty" state isn't it?
@tbfleming  any Redux Black Magic to dispatch GCODE_DIRTY on any others reducers change?

tbfleming, 2017-02-23:
@jorgerobles : the gcode reducer can listen for all actions which begin with DOCUMENT_ or OPERATION_. It will have to ignore changes to document.selected, document.visible, operation.expanded, and operation._docs_visible.

jorgerobles, 2017-02-23:
Hmm  @tbfleming  It can watch for other events but the state is scoped to GCODE. So cannot ignore specific changes :| (as far as I know)

tbfleming, 2017-02-23:
The gcode reducer decides what actions it does and does not listen to.

jorgerobles, 2017-02-23:
Yes, that is correct, but  `state` variable is not loaded with all the state, only with gcode state.
I hope calling getState from action obtains the preprocesed overall state, if not, I have to move gcode reducer last on the combineReducers list

tbfleming, 2017-02-23:
The actions have everything you need; you don't need anything in the document state or the operation state.
* if action.type begins with DOCUMENT_ or OPERATION_, but is not DOCUMENT_SET_ATTRS or OPERATION_SET_ATTRS then light the button
* if action.type is DOCUMENT_SET_ATTRS and any field is present that is not 'selected' or 'visible' then light the button
* if action.type is OPERATION_SET_ATTRS and any field is present that is not 'expanded' or '_docs_visible' then light the button

jorgerobles, 2017-02-23:
Ook, I understand now. I need a nap.

cprezzi, 2017-02-23:
This issue is probably related to #170.

jorgerobles, 2017-02-26:
This is done and live!

closed 2017-02-26 by jorgerobles