LaserWeb/LaserWeb4 #168: Log

donkjr, 2017-02-23:
I previously suggested a log be created of commands being sent to the target.
If I recall that suggestion was accepted, maybe I was suffering a disillusion.
Is this suggestion still being considered for LW4 or was it rejected?

The Gcode export and view functions in LW4 is an awesome help for understanding what WILL be sent to the target. There are times during troubleshooting when I need to know what is actually BEING sent to the target. 
In LW3 I set up a batch (crude and ugly) to capture such so I am just wondering how I would implement something similar in LW4.

cprezzi, 2017-02-23:
There is a logfile option on the server side (in lw.comm-server), which can be activated in the config.js.
`config.logLevel = x;` where 0=no_logging and up to 3=max_details.

cprezzi, 2017-02-23:
An existing logfile.txt is overwritten on server start.

closed 2017-02-28 by openhardwarecoza