LaserWeb/LaserWeb4 #169: if (served by electron) {catch f5 and f12}

openhardwarecoza, 2017-02-23:
Sorry, but I need this in lw4 to make the packager work optimally (: 

May need to wrap up all the below in a function that checks if we are running from electron, or somewhere else.
In LW3-Electron we used 'if (require) { do electron stuff here}' - not sure if thats the perfectly correct way though

Function to bind F5 as refresh and F12 as open devtools:  After being made an EXE standard keyboard shortcuts doesnt work as electron != chrome

```
document.addEventListener('keydown', function(evt) {
        if (evt.which === 123) {
            try {
                var focusedWindow = require('electron').remote.getCurrentWindow();
                if (focusedWindow.isDevToolsOpened()) {
                    focusedWindow.closeDevTools();
                } else {
                    focusedWindow.openDevTools();
                }
            } catch (error) {
                console.warn(error);
            }
        } else if (evt.which === 116) {
            location.reload();
        }
   });
```

Originally from @copet80 's work https://github.com/LaserWeb/LaserWeb3/commit/2e6531341ac04863d35521ee6b9cf0d1d96827fa

jorgerobles, 2017-02-26:
should (F12) work for  end user electron? Or only for developers?

openhardwarecoza, 2017-02-27:
I'd say users too. There will be times where a user reports something not
working and we can ask them to go to console to check / clear LocalStorage
/ etc

jorgerobles, 2017-02-28:
@openhardwarecoza, I've integrated the code. Please build and check.

openhardwarecoza, 2017-02-28:
Its catching F5 (refresh)just fine (also catches F11=fullscreen which is cool) 
But nothing happens on F12
I also checked with devTools enabled and nothing logged (no error)

openhardwarecoza, 2017-02-28:
Whats funny is that in webpack-dev-server (ie before building with electron) it seem to launch and close devtools... weird

jorgerobles, 2017-02-28:
I think the integration should be done in the code that builds electron, overloading a window.isElectron or so. We got some warnings of compilation when running on browser.
BTW. F12 is mapped on dashboard of OSX, so never work for me :(

openhardwarecoza, 2017-02-28:
Ahh, maybe F12 isnt the best choice?

jorgerobles, 2017-02-28:
What if we add two tiny icons on the bottom of Dock, one for reload and one
for Devtools?

openhardwarecoza, 2017-02-28:
> issuecomment-283061736

I'd be happy with that too! Maybe even hide them on the settings tab?

jorgerobles, 2017-02-28:
I will do!
Some caveat: I'm going to use this kind of condition (below)
`getFocusedWindow` should be implemented on Electron build, so it abstracts the require electron and get Focused window calls.
This way, will compile ok for browser.

<pre>
if (window.getFocusedWindow){
                var focusedWindow = window.getFocusedWindow();
               ...
            }
</pre>

jorgerobles, 2017-02-28:
[image]

openhardwarecoza, 2017-02-28:
Love it

openhardwarecoza, 2017-02-28:
Reload works

But alas, Open DevTools says it's my Pal, but still refuses to do me any favours :P

jorgerobles, 2017-02-28:
LOL, yes. but that is because there's no window.getFocusedWindow. No window, no favours.

openhardwarecoza, 2017-02-28:
window: 
global {speechSynthesis: SpeechSynthesis, caches: CacheStorage, localStorage: Storage, sessionStorage: Storage, webkitStorageInfo: DeprecatedStorageInfo…}

window.getFocusedWindow:
undefined

Well, thats in Electron, it has window.... But it doesnt have window.getFocusedWindow

openhardwarecoza, 2017-02-28:
(github fails to upload any screenshots for me)

openhardwarecoza, 2017-02-28:
http://pasteboard.co/DS8XJwT30.png

openhardwarecoza, 2017-02-28:
const remote = require('electron').remote;
const app = remote.require('electron');
app.BrowserWindow.getFocusedWindow()

jorgerobles, 2017-02-28:
hmm could define getFocusedWindow() at top level?
<pre>
function getFocusedWindow(){
   const remote = require('electron').remote;
   const app = remote.require('electron');
   return  app.BrowserWindow.getFocusedWindow()
}
</pre>

jorgerobles, 2017-02-28:
even window. getFocusedWindow = () => {...}

openhardwarecoza, 2017-02-28:
Sorry made an edit after your comment

Thats where I am still not sure in reactworld

jorgerobles, 2017-02-28:
Just try 
<pre>
window.getFocusedWindow = function(){
   const remote = require('electron').remote;
   const app = remote.require('electron');
   return  app.BrowserWindow.getFocusedWindow()
}
</pre>

openhardwarecoza, 2017-02-28:
http://pasteboard.co/DSo3AmjA4.png

openhardwarecoza, 2017-02-28:
After defining the function, the button works beautifully

jorgerobles, 2017-03-01:
@openhardwarecoza Could you merge the buttons into the dev_comms to get a debugger build?

*slighty complain* I've got a bunch of confusing commits after comms integration

openhardwarecoza, 2017-03-01:
Well, technically we can use dev_es6 instead of devcomms now in electron.
Will build this afternoon

Yeah thats why i always just work in 'master' - merging branches makes me
more confused than just handling conflicts and changes one by one. I know
its wrong but it works

tbfleming, 2017-03-01:
@openhardwarecoza it's not wrong. I only prefer branches when intermediate changes break builds.

jorgerobles, 2017-03-04:
@openhardwarecoza I've been digging the docs. Seems we have to use access to ipcRenderer to do messaging between the Laserweb (webcontents) and Electron (mainProcess).
Main issue is 
 https://github.com/LaserWeb/LaserWeb4/issues/169#issuecomment-283127415 
produces compilation errors on webpack.

I was fiddling a workaround, specifiying a `  "bundle-electron":"webpack --progress --colors --config=webpack.config-electron.js --target=electron-renderer",`
that makes use a different laserweb `index-electron.js` that requires `electron` module.

... but the I got a ****** hairball.

openhardwarecoza, 2017-03-05:
I have this kinda working:

In https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/components/settings.js#L335 I replaced the function with 

```
handleDevTools:() => {
            const electron = window.require('electron');
            const app = electron.remote;
            var focusedWindow = app.BrowserWindow.getFocusedWindow()
            if (app.BrowserWindow.getFocusedWindow){
                if (focusedWindow.isDevToolsOpened()) {
                    focusedWindow.closeDevTools();
                } else {
                    focusedWindow.openDevTools();
                }
            } else {
                console.warn("Can't do that, pal")
            }
        },
```

That works beautifully in Electron now, and doesnt give a compile error in Webpack either.

In the Browser, I get an error when clicking Open DevTools.  Maybe we can extend the Render function to also check  
```
   if (app.BrowserWindow.getFocusedWindow){
``` 

and not display the Open Devtools button in the Browser?

jorgerobles, 2017-03-05:
Push it, I will check :)

openhardwarecoza, 2017-03-05:
### electron ###

[image]

### browser ###
[image]

openhardwarecoza, 2017-03-05:
I added the 'if' in render() 
I renamed the button to Toggle instead of Open (since it toggles it closed too)

jorgerobles, 2017-03-05:
Perfect!! One less to go!

closed 2017-03-05 by openhardwarecoza