LaserWeb/LaserWeb4 #170: Raster Simulation Rendering

donkjr, 2017-02-23:
[image]

LW4 is a huge improvement over LW3 and I am appreciative of all the volunteer work that has gone into it. I did not contribute to the code and have paid nothing for the use of this program so I have no right to criticism. Therefore the below is not intended as a criticism of the design, rather one users perspective of its operation. 

1. Gcode is allowed to be generated without all the required Raster definition pane fields being completed. If you Gcode/generate without the minimal fields completed the g-code is generated but the simulation (using  the simulator slider) will not animate  properly. My tinkering revealed that as a min the cut speed and # passes are required for proper simulation animation.
2. If the image is removed via killing the Raster Definitions or the "Document" list the raster simulation overlay remains (by use of the simulator bar). The only way to remove the simulation overlay is to refresh the browser.
-------
a: As a raster engraving designer I would like an indication if a Raster definition pane is complete or not and inhibit the generation of Gcode and simulation until it is.
b: As a raster engraving designer I would like the associated simulation animation to be removed when I remove a Raster definition pane.

tbfleming, 2017-02-23:
a: I agree

b: I disagree. The simulator shows gcode, nothing else. It's confusing in this case because of a. The rasterizer produces an insanely large number of passes when the values are bad. The simulation doesn't disappear when you remove an operation because the gcode doesn't change. Only the Gcode Generate button and Gcode Load button ever change gcode.

cprezzi, 2017-02-23:
What if we dasable the generate gcode button until all params are filled. And disable the Simulator until there is gcode?

cprezzi, 2017-02-23:
@donkjr How big and red the hints for missing params must be to receive attention? ;)

donkjr, 2017-02-23:
@cprezzi 
Heh, I guess if the red hints covered up the Gcode button it would have prevented my stupidity :). 
I get your point. For whatever reason I was blinded by the GCODE button.

I have this notion that a software UI, if at all possible, should not let you do stupid or newbie stuff ....... 
It seems that if its smart enough to flag me to fill it in its smart enough to do blank and validity checks. 
I know it is doing validity checks because when I enter out of bound values the flag stays there until it is correct ...
I think controlling when the Gcode button is active (as discussed above) fixes most of this confusion.

donkjr, 2017-02-23:
@tbfleming 
Ok if I understand this correctly;  if the Gcode still exists after the image definition is removed then how does one clear out stored Gcode without doing a browser refresh?

tbfleming, 2017-02-23:
@donkjr If you click the Generate button when there are no operations it will generate empty gcode.

tbfleming, 2017-02-23:
@donkjr Or, you could hide it using the checkboxes below the workspace.

donkjr, 2017-02-23:
@tbfleming Maybe a "clear" button up on the Gcode ribbon :).

tbfleming, 2017-02-23:
A clear button sounds nice.

jorgerobles, 2017-02-23:
Related to #100

closed 2017-02-28 by jorgerobles