LaserWeb/LaserWeb4 #175: LW4 Does not load due to webcam on OS X

maudette01432, 2017-02-27:
@ https://laserweb.github.io/LaserWeb4/dist/

Javascript console shows:

```
Uncaught Error: Cannot find module "getusermedia"
    at webpackMissingModule (webcam.js:11)
    at Object.<anonymous> (webcam.js:11)
    at __webpack_require__ (bootstrap b08eee3…:555)
    at fn (bootstrap b08eee3…:86)
    at Object.<anonymous> (settings.js:22)
    at __webpack_require__ (bootstrap b08eee3…:555)
    at fn (bootstrap b08eee3…:86)
    at Object.<anonymous> (dock.js:16)
    at __webpack_require__ (bootstrap b08eee3…:555)
    at fn (bootstrap b08eee3…:86)
```

OSX 10.12.3/Chrome 56.0.2924.87 (latest)

jorgerobles, 2017-02-27:
@openhardwarecoza  could you checkout current version and run npm run installdev, and then golive? I think is a compilation issue. I'm away from my devel desktop :)

openhardwarecoza, 2017-02-27:
On it (: let me do so

openhardwarecoza, 2017-02-27:
Done! Was indeed missing the installdev (:

[image]

closed 2017-02-27 by openhardwarecoza