LaserWeb/LaserWeb4 #176: Undo function scope

jorgerobles, 2017-02-28:
Hey all. 

the UNDO function is working since few months, but I realize some events should not be captured with the UNDO, as tends LW to go nuts.

@tbfleming Could translating documents on workspace send only one event per StartDrag/StopDrag couples, also camera movements (if they are considered to be captured for undo.)

Please all, take a while to UI, with devtools opened, and check what events should/should not be captured for undo, and post here.

tbfleming, 2017-02-28:
@jorgerobles the display is based on the current state, so it needs the repeated actions. Most (all?) number fields also produce repeated actions (hold down the up or down arrow key when a number field has focus), so we probably need a general solution.

How about this: the undo system could store the last action received. It should clear it whenever there's an undo or redo. If the undo system receives an action that matches (see below) the currently-stored one then don't change the undo history.

Two actions match if:
* They have the same type, and
* The type is *_SET_ATTRS, and
* The set of attributes changed by the actions have the same names

jorgerobles, 2017-02-28:
@tbfleming I've done the way you suggested and catched some Undo bugs. I think is going well enough. Thanks!

DarklyLabs, 2017-02-28:
@jorgerobles The most common time we have needed undo is when we have accidentally moved a vector and want to put it back to exactly where it was.
Currently it means re-loading the document again and starting to setup the operations from scratch. 

We have tried ctrl-z recently (with fingers crossed) but is hasn't worked. I guess this is not being captured.

Something I would recommend if you are considering being able to undo multiple steps is to place a limit on this. Perhaps the last 5 operations are undoable. I have seen many graphical programs become very unstable when adopting infinite undos.

jorgerobles, 2017-03-01:
@DarklyLabs DOCUMENT_SELECT_TRANSLATE was catched indeed (was before but not as cool as now!)
I've added a 2s timeout, so if you move an element, and keep moving within 2 seconds time range, only first move counts. If you move, wait 2s and then move, both movements are captured on Undo.

Undo is limited to 10 snapshots back.

DarklyLabs, 2017-03-01:
@jorgerobles Not sure what i'm missing but undo is not working for me on the latest dev-es6. Does it need to be activated?

DarklyLabs, 2017-03-01:
@jorgerobles Ok, looks like it's ctrl-z on a mac as well (normally cmd-z is standard)

At the moment if I drag a document 100mm to the right, continually pressing undo will gradually move it closer to the original position but it doesnt get there because there is a limit of 10 undos.

jorgerobles, 2017-03-01:
I will try to catch Meta (cmd or ctrl) instead ctrl.
If you are using dist, maybe I forgot to push to gh-pages

jorgerobles, 2017-03-01:
@DarklyLabs should be working now. It catches ctrl+z and cmd+z as undo. Also fixed some weird component scope keydown bug.

DarklyLabs, 2017-03-01:
@jorgerobles Excellent. Just tested dev-es6 and it looks like both cmd-z and the undo position problem is resolved. We will test more but well done.....again!

closed 2017-03-02 by jorgerobles