LaserWeb/LaserWeb4 #178: Save settings / profiles / app snapshot not working in electron app 4.0.36

cojarbi, 2017-03-01:
Im not sure this is a bug, most likely not coded yet since the save action maybe its not the same for the app. Load does work.

I could not test in 4.0.40 as i dont have a build.

DarklyLabs, 2017-03-01:
Tested in 4.0.40 and the save feature does work.
The one bug I can see is that if 'cancel' is selected when saving then a null.json file is still saved.

cojarbi, 2017-03-01:
@DarklyLabs is this on an electron app?

DarklyLabs, 2017-03-01:
@cojarbi No, we are working from dev-es6 for now.

cojarbi, 2017-03-01:
@DarklyLabs i suspect this is an issue with the electron desktop app only. I have tried dev-es6 4.0.40 and it also works for me

DarklyLabs, 2017-03-01:
Just including @jorgerobles on this.

jorgerobles, 2017-03-01:
I'll check ASAP

jorgerobles, 2017-03-01:
@openhardwarecoza please add this an [ELECTRON] tag.

openhardwarecoza, 2017-03-01:
done.

openhardwarecoza, 2017-03-05:
@tbfleming @jorgerobles 
Prompt is not supported under electron, thats why these doesnt work. We will have to replace prompt() where its used with something that is supported in electron

openhardwarecoza, 2017-03-05:
https://github.com/electron/electron/issues/472

jorgerobles, 2017-03-05:
> electron/electron#472 <https://github.com/electron/electron/issues/472>

Ok, will find a polyfill.

jorgerobles, 2017-03-05:
https://github.com/konsumer/electron-prompt?files=1

jorgerobles, 2017-03-05:
Added npm dialog for electron with fallback on native dialogs on browser. Please check

jorgerobles, 2017-03-05:
Indeed thanks to @cojarbi We have found worse errors. Cannot overload Window. several "services" are bound to window (window.videoCapture, that handles webcam streams) but are not found on electron builds.

DarklyLabs, 2017-03-06:
@jorgerobles You probably already are aware but the save button is still not functional in electron 4042.

jorgerobles, 2017-03-06:
@DarklyLabs Yes I am, thanks. I did a couple of comments before yours, I was expecting to solve them easily but not. Anyways, native chrome prompts and confirms are not working. That's the reason it's failing, and so I'm looking solutions.

jorgerobles, 2017-03-07:
@DarklyLabs should work now with latest compile.

DarklyLabs, 2017-03-07:
@jorgerobles Compiled for mac successfully.
Save button now working. Even opens up a dialog box to choose where to save to now. Yeah! Well done.
Settings tab is working again.
Machine and GCode tabs are missing some headings above text areas.

Having some trouble compiling for windows. Will keep at it. Perhaps someone out there could verify if windows is compiling correctly for them?

openhardwarecoza, 2017-03-07:
Just compiled on Windows, without issue

cd ../LaserWeb4
git pull
npm run installdev (to grab new modules and new lw.machines)
cd ../lw.comm-server
git pull
npm install (just in case)
npm run dist

jorgerobles, 2017-03-07:
:)

closed 2017-03-07 by jorgerobles