LaserWeb/LaserWeb4 #180: Electron build number not updated on the top border of the app

cojarbi, 2017-03-01:
Its suppose to be 4.0.36

[image]

openhardwarecoza, 2017-03-01:
Needs a manual edit.

cprezzi, 2017-03-02:
I my numbering scheme with 4.0.x even correct? Or should it be 0.4.x?

openhardwarecoza, 2017-03-02:
@lautr3k took us back to 0.4.0 with the start of the lw4 backend dev
(; LW3s bad stability seemed more like a 0.3.0 lol.

I stand by it being the 4th iteration so I'd say 4.x.x

But since both frontend, backend, and electron are seperate, it does pose
an interesting challenge of how to struct a version then

cprezzi, 2017-03-02:
It's easy to add a command to the server for the app to query for the server version. I've planned that anyways. So we could show the app and server versions separately on the about window.

cprezzi, 2017-03-02:
I have added the frontend and backend versions to the about tab and removed the hardcoded version number from the browser title.

openhardwarecoza, 2017-04-17:
Thanks @jorgerobles you accidentally fixed #180 too with https://github.com/LaserWeb/LaserWeb4/issues/259 's fixes (;

[image]

closed 2017-04-17 by openhardwarecoza