LaserWeb/LaserWeb4 #181: Laser Fill Path speed

DarklyLabs, 2017-03-01:
We are noticing some SVG files take significantly longer to generate fill paths than others.

The attached file sits on 0% gcode generation for about 1 minute and then completes.  Some other SVG files take even longer. We have examined the SVG files and cannot see anything that could be causing this slowdown.

Perhaps someone may have an idea?

[twirl.svg.zip](https://github.com/LaserWeb/LaserWeb4/files/812272/twirl.svg.zip)

tbfleming, 2017-03-02:
Here are things which impact speed:
* Number of points after curve-to-line conversion (lw.svg-parser controls this)
* Path complexity
* Number of raster lines it takes to cover the path

Path complexity is hard to measure and harder to explain, but plays a huge role. Here's how part of it plays out with this SVG:

[image]

The CAM's polygon operations rely on a scan-line algorithm. The scan line (blue line above) is always vertical. The raster lines (black) are under user control. Time goes up with:
* The number of path-to-path and path-to-raster intersections
* The number of intersections each scan line makes with:
  * The path
  * The raster lines
* The number of scan lines. There's a scan line for each unique X coordinate after curve-to-line conversion.
* Other aspects that I forgot

DarklyLabs, 2017-03-02:
@tbfleming Thanks for the in-depth explanation.
What we have been noticing is that this very long pause before anything appears to be happening is causing users to assume something has gone wrong. 
Would there be a way to report progress while this calculation process is taking place and display the in the generation progress bar?

tbfleming, 2017-03-02:
The best we could hope for is replacing the number with a sequence of named steps. Unfortunately most steps don't have a good way to measure progress; they're done when they're done and they can't predict when.

DarklyLabs, 2017-03-03:
The design above took 2min 20sec to generate the fill path.
0.1 line width, 100x100mm

[image]

DarklyLabs, 2017-03-03:
This design took 5min 15sec to generate fill path.
0.1 line width, 100x123mm

[image]

DarklyLabs, 2017-03-03:
@tbfleming Perhaps there are some speed optimisations possible to help reduce these times?
The raster gcode generation is lightning fast in comparison.

tbfleming, 2017-03-03:
There probably are potential optimizations.

jorgerobles, 2017-03-03:
@tbfleming I think the lag is on preflight (examineDocTree)

cprezzi, 2017-03-03:
What about converting the vector data to a raster graphic and create the gcode for the fill operation with the raster gcode generator? Coud that speed up the process?

maudette01432, 2017-03-03:
@cprezzi I was thinking the same thing.  The current "line fill" isn't quite a raster.  I've seen it jump around in places on me...lowering efficiency.

jorgerobles, 2017-03-03:
@cprezzi convert to bitmap is a fair solution, We could even propose as a new "tool", thus can be easier in the design tool of choice.

tbfleming, 2017-03-03:
Line fill isn't meant to be a raster. There are cases where raster will produce nicer results and cases where line fill will produce nicer results. Adding the ability to convert paths to raster so users can choose would be nice.

jorgerobles, 2017-03-04:
@tbfleming, I've been thinking the paths needed to convert to raster are the result of computing laser fill if not it will show hollow shapes. So the computing issue won't be resolved :|

@DarklyLabs better just raster the complex shapes on your design tool and keep vectors for outline finish.

cprezzi, 2017-03-04:
@jorgerobles If the vector data has fill color (or you set it to) and is posted to a canvas, then you will get a raster picture with infill for the raster process.

tbfleming, 2017-03-04:
> @tbfleming I think the lag is on preflight (examineDocTree)

Dang. That's the worst spot because usually examineDocTree spends most of its time in Clipper. I was hoping it was after that point.

I just tried to verify the location that but couldn't: ```Uncaught Error: Cannot find module "socket.io-client"```.

jorgerobles, 2017-03-04:
run `npm run installdev` dev-es6 has been merged with dev_comms

tbfleming, 2017-03-04:
@jorgerobles I did.

jorgerobles, 2017-03-04:
@tbfleming I've add progress on Preflight, as far as elegant I could. 
I've narrowed a lag upon Gcode representation. It's computed fairly well, but took +6min to show on workspace.

[workspace_fill_lag.json.txt](https://github.com/LaserWeb/LaserWeb4/files/818931/workspace_fill_lag.json.txt)

cojarbi, 2017-03-04:
@jorgerobles im unable to generate anything after this.

jorgerobles, 2017-03-04:
I've tested @DarklyLabs file and I have a 1m20s lag on Preflight computing.

Workspace: [twirl2-workspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/818938/twirl2-workspace.json.txt)

@DarklyLabs same file opened on AI and breaking compound object, lasts half to generate.
[twirl1.svg.zip](https://github.com/LaserWeb/LaserWeb4/files/818941/twirl1.svg.zip)

jorgerobles, 2017-03-04:
@cojarbi does it throw any debug error?

cojarbi, 2017-03-04:
@jorgerobles im not getting anything.

[image]

cojarbi, 2017-03-04:
Also getting this on the terminal while npm start.
[image]

DarklyLabs, 2017-03-05:
@jorgerobles Tested the new version wit pre-flight in progress bar. It moves to 50% very quickly and then waits for completion. This is still a much better result than previous as it shows users the process is happening.

We tested the twirl with compound paths removed and achieved exactly the same processing time (2min20sec). It did import the svg file faster though.

tbfleming, 2017-03-05:
Missing module fixed in  70801d5

jorgerobles, 2017-03-05:
@cojarbi, the npm start log is correct. Please attach the files that throw call stack problems.

cprezzi, 2017-03-05:
@tbfleming That module is usualy not needed to be add manually, as node server serves it by default. Did you start the server too? 

But what I found: Every time I update lw.comm-server, I need to delete lw-comm-server folder in lw4's node_modules subfolder and do the installdev again. It's not updated automatically. What did I wrong?

cojarbi, 2017-03-05:
@jorgerobles here is the file from above. I think the problem is it has a gazillion of path not joined. If that the case and this could happen to other users thus creating support issues can we get an error to display " go fix your file " ?

If i join all the paths it works

[0001.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/819368/0001.svg.txt)

tbfleming, 2017-03-05:
> But what I found: Every time I update lw.comm-server, I need to delete lw-comm-server folder in lw4's node_modules subfolder and do the installdev again. It's not updated automatically. What did I wrong?

Does it update if you run installdev without deleting it?

cprezzi, 2017-03-06:
@tbfleming No, it doesn't.

jorgerobles, 2017-03-07:
@cojarbi it's a neat idea but have to figure how to do that.

DarklyLabs, 2017-04-01:
Just updated the tags on this. So far the processing speed of vector fills and general interactivity slowdowns after loading an average-complexity vector document has been the most discussed short fall from our users.

domenic-d, 2017-07-15:
@jorgerobles @tbfleming 
I just ran some tests with the display cache and fill operations:

Build: 4.0.989
OSX

Display Cache = off, Threads = 2
Completion time = 3min 22 sec

Display Cache = off, Threads = 4
Completion time = 4min 24sec

Display Cache = on, Threads: 2 or 4
Completion time = 1min 9sec

Why does the display cache have such a big impact on the fill performance?

jorgerobles, 2017-07-15:
Display cache stops all webgl rendering until an event is triggered over workspace, dropping gpu consumption. I think webgl in certain cases is not using gpu but cpu, despite the chrome settings. That could be a os/chrome related issue

jorgerobles, 2017-07-15:
Also, have to say, the threading system is not really concurrent in the same operation, but on several operations.

tbfleming, 2017-07-15:
I suspect @jorgerobles is right.

domenic-d, 2018-01-22:
Closed as it's as fast as it's going to get.

nipunawg, 2019-01-10:
I might have found a workaround. i had an SVG taking around 10m to finish generating gcode. I tried loading it in inkscape and converting all curves into very short straight line segments (select all nodes in all the paths, click "add nodes" 3 or so times, convert to straight line segments, save) inkscape lags a bit in the process but only for less than a minute and the whole process takes about a minute.
it brought my gcode generation time down to 40s for that file with no loss in curve quality!

cprezzi, 2019-01-10:
@nipunawg The same result you could proably reach by just setting our option `GCODE CURVE LINEARIZATION FACTOR` (under in LW settings->gcode) to a value like 0.5 or more.