LaserWeb/LaserWeb4 #183: Japaneses laser

FabCreator, 2017-03-02:
FABOOL are a new laser cutter which have an app, which i think may be opensource,
https://github.com/arii2016/FaboolApp
Might be some use.

FabCreator, 2017-03-02:
http://www.smartdiys.cc/manual/fabool-laser-co2-howtouse-software/ 
http://www.smartdiys.cc/fabool-laser-mini/

maudette01432, 2017-03-02:
Not an issue...

FabCreator, 2017-03-02:
@maudette01432 erm, as far as i understand the way we have been using Github not every Issue is an Issue some are ideas feature requests or links to other projects in the past?

cprezzi, 2017-03-02:
Hm. Until now, all issues & discussions was about our development, not alternative products. I think G+ would be the better place to post something like this.

maudette01432, 2017-03-02:
Yeah I think the message is a little confused with issues like "Lets seal from the Chinese (;" https://github.com/LaserWeb/LaserWeb4/issues/155" - I don't want to be a police officer..but we should have consistency.  I agree with Claudio, this along with issue #155 belong more in the G+ discussion group.  Wile some of the code in the repo above may add some value it's not related to a LW4.  I'm asking Peter/Anthony to chime in here.  I think we should probably close this and #155.

On a side note the software linked in the second post does not work in chrome on OS X.  It fails to load.

cprezzi, 2017-03-02:
Right. It's a bit fuzzy what we did in the past ;) 
The advantage to having it here is, that it doesn't get lost in the G+ history. 
It shouldn't be too disturbing if we mark it with the discussion tag.

FabCreator, 2017-03-02:
The lets steal from the chinses is why i posted it :P these guys have there own g code generator and say they are opensource it also appear to run in the web, so there may be things in there you can merge with lw :) thats why i linked it.
G+ is a pain to store info etc. Maybe we should start a LW forum instead? i just made a discourse for our users. could do the same for LW.

maudette01432, 2017-03-02:
LOL.  This was stolen from the Lasersaur project.  http://www.lasersaur.com  - Check out 

https://github.com/arii2016/FaboolApp/blob/master/frontend/img/lasersaur-dino-lasaurapp-190.png

Which the original application repo can be found here:
https://github.com/stefanix/LasaurApp

Might be easier to read than the Japanese comments....it's also a little stale coming from a couple years ago...but there might be some good code there.

closed 2017-03-07 by openhardwarecoza