LaserWeb/LaserWeb4 #185: Add Emblaser Support community

DarklyLabs, 2017-03-04:
To help manage questions from Emblaser users, can we have out support community added to the 'about' section?

**Emblaser Support Community**
[](https://darklylabs.zendesk.com/hc/en-us/community/topics/115000181507-Laserweb)

FabCreator, 2017-03-06:
Perhaps this can modify you select the machine profile,
This way we could link all machines to respective communities be it cohesion 3d, smoothie, GRBL, fabcreator, k40 etc

cprezzi, 2017-03-06:
@FabCreator The idea with the machine presets is much better than a fixed entry. As LaserWeb is open for all machine or controller supplyers, it wouldn't be ok to have just one of them in the links.

jorgerobles, 2017-03-06:
I propose to add a new setting, not editable on UI, called machineAbout.
This about should be filled with the name of the md file with the same name. Ex: 

- Fabcreator: fabcreator.md
- Emblaser2: emblaser2.md

This will be rendered on the about pane, before credits. Just compose the file and PR to https://github.com/LaserWeb/lw.machines

jorgerobles, 2017-03-06:
Done! @DarklyLabs check the update done to lw.machines (fixed some files, and added emblaser2.md)
@FabCreator please PR your settings :)

DarklyLabs, 2017-03-06:
@jorgerobles The link looks correct. Perfect.
I do see some things that need fixing in our machine profile. Mainly the com ports that are listed in there. We should remove them as they will be different on users's machines. I didn't realise they were saved along with the machine profile.

FabCreator, 2017-03-06:
@jorgerobles is it possible to add macros? also how easy is it to update the profile?

jorgerobles, 2017-03-06:
@fabcreator Currently macros are not bound to machine profiles, but could
work on in if proven necessary.

Making a machine profile is just as easy as customize your settings and
then go to machine profiles on settings top pane, and save.

Then export with settings->tools->machine profiles->save

FabCreator, 2017-03-06:
Ah okay, @jorgerobles it would be great if we could add marcos to the control panel based on profile as we have a few features.
Home material
Move to origin
Compressor On / Off (used to be a button in lw2 i think)
Alignment test/setup/testfire
I have added our config to the file, could you check i did it correctly, thanks :)

cprezzi, 2017-03-06:
@FabCreator You don't need a alignment testfire marco. That what's whout the "Laser Test" button is for. You just have to set the desired power and duration in the settigs gcode tab.

cprezzi, 2017-03-06:
If you set the duration to 0, the button acts as a toggle on/off button (showing a red outline while on).

cprezzi, 2017-03-06:
@jorgerobles The actual preset marcos (set zero, laser tracer, laser off) are not needed as they are already available though buttons on the jog tab, but a `Goto XY Zero` could be usefull.

jorgerobles, 2017-03-06:
@cprezzi bundled macros are on data/macros.json feel free to add or remove
whenever necessary :)

FabCreator, 2017-03-07:
@cprezzi we never managed to get test fire to work in laserweb 3, ill try it in Laserweb4.
Our machines require m3 and m5 i dont know if this happens with the test fire?

cprezzi, 2017-03-07:
@FabCreator The code behind the "Laser Test" button uses firmware specific (g)code for firing the laser. On smoothie, we use the fire command. You only need to config the power and duration values in settings, that should be all.

cprezzi, 2017-03-07:
M3 would only fire the laser when combined with a G1-G3 (because of the adaptive laser power feature).

FabCreator, 2017-03-07:
okay i will run some tests, then. :)

DarklyLabs, 2017-03-10:
As far as we can tell, @jorgerobles has resolved the original issue with respect to 'About content based on machine profile'
I suggest the macro issue be moved to a new issue. i will close this one.

closed 2017-03-10 by DarklyLabs